tuya-cloudcutter/tuya-cloudcutter #659: hostapd_free_hapd_data: Interface wlan0 wasn't started

mocarela, 2024-05-03:
I'm trying to cut off from the cloud my "tuya-generic-s18-ir-repeater", but failing on the last step (at least as far as I understand the whole thing).

```
~/tuya-cloudcutter# ./tuya-cloudcutter.sh -r -w wlan0
Building cloudcutter docker image
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  3.216MB
Step 1/10 : FROM python:3.9.18-slim-bullseye AS base
 ---> 83b8dddc8bfb
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 8e5f6ae9ef90
Step 3/10 : FROM base AS python-deps
 ---> 8e5f6ae9ef90
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 4358d26ea583
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> fd92944fc2e9
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> b80db5632f67
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 1288d75a31b2
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 1288d75a31b2
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 31d655193e9a
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 62b6c2b28d9c
Successfully built 62b6c2b28d9c
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 1
Detaching requires an SSID and Password, please enter each at the following prompt
In order to provide secure logging, the values you type for your password will not show on screen
If you make a mistake, you can run the detach process again
You can also pass credentials via the -s command line option, see './tuya-cloudcutter.sh -h' for more information'
Please enter your SSID: myaccesspoint
Please enter your Password:
Loading options, please wait...
[?] How do you want to choose the device?: From device-profiles (i.e. custom profile)
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-s18-ir-repeater
 > tuya-generic-s18-ir-repeater

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: tuya-generic-s18-ir-repeater
Selected Profile: oem-bk7231n-irbox-mol-ty-2.0.0-sdk-2.3.1-40.00

[... 186 lines cut ...]
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 59 T2  10h30m
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  3 router  10.42.42.1
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 client provides name: wlan0
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 DHCPREQUEST(wlan0) 10.42.42.32 18:de:50:bc:fc:52
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 tags: wlan0
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 DHCPACK(wlan0) 10.42.42.32 18:de:50:bc:fc:52 wlan0
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 next server: 10.42.42.1
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  1 option: 53 message-type  5
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 51 lease-time  12h
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 58 T1  6h
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 59 T2  10h30m
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  3 router  10.42.42.1
May  3 15:27:15 dnsmasq-dhcp[16]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
[15:27:16.690331 MQTT] Connected
```

Any idea what's wrong in my case?

Cossid, 2024-05-03:
Your device is having issues finishing the DHCP request in time and gets stuck.  You can restart the network join process;

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

mocarela, 2024-05-04:
Thanks for the help... it worked! I should have read that. :$

May I ask another question, since I'm quite new in this IoT universe. After I have cut it off, it still works with the Tuya fw, thus I have to use kind of LocalTuya integration in HA right? And because it looks it doesn't support it, I will have to flush it also.

What fw do you recommend? I would just need to learn IR codes (of other remotes) and transmit them (either manually or with automations). Beside included ESPHome and OpenBeken, do I have any other alternatives that can be flashed OTA using tuya-cloudcutter?

Cossid, 2024-05-04:
> After I have cut it off, it still works with the Tuya fw, thus I have to use kind of LocalTuya integration in HA right?

Yes, you would use LocalTuya for integration into HA.

> And because it looks it doesn't support it, I will have to flush it also.

I don't understand what you're asking here.  You could possibly open an issue with LocalTuya to ask for support for your devices.  I'm not sure what you mean by flush it.

> What fw do you recommend?

Personally, I mostly use ESPHome, but it does have a high learning curve.  I know both have some form of IR support, but don't have much experience with IR on either yet.

> Beside included ESPHome and OpenBeken, do I have any other alternatives that can be flashed OTA using tuya-cloudcutter?

I'm not aware of any other firmware that supports Beken yet.

closed 2024-05-04 by Cossid

mocarela, 2024-05-05:
> > And because it looks it doesn't support it, I will have to flush it also.
> I don't understand what you're asking here. You could possibly open an issue with LocalTuya to ask for support for your devices. I'm not sure what you mean by flush it.

I meant flash it, since local tuya doesn't support it.

> > Beside included ESPHome and OpenBeken, do I have any other alternatives that can be flashed OTA using tuya-cloudcutter?
> I'm not aware of any other firmware that supports Beken yet.

Does ESPHome support it at all?

Cossid, 2024-05-05:
> Does ESPHome support it at all?

Yes, ESPHome supports Beken at the core level.  IR seems to be mostly working, though there may be some bugs remaining, I'm not certain, I haven't followed the latest changes there.