tuya-cloudcutter/tuya-cloudcutter #661: Could not connect to SSID

bigboigahoy, 2024-05-05:
I have several Merkury bulbs (MI-BWXXX) that seem to loop during the AP step. Below is from a Merkury MI-BW904 on v1.0.1

`Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
......
Found access point name: "Merkury-5D5E", trying to connect...
Device 'wlan0' successfully activated with '9ad44e5a-7b66-40d6-8d4b-9a9af8b11a26'.
Wiping NetworkManager configs
..
Found access point name: "Merkury-5D5E", trying to connect...
Device 'wlan0' successfully activated with 'ce25e59a-da96-4a39-98b9-25914652566c'.
Wiping NetworkManager configs
..
Found access point name: "Merkury-5D5E", trying to connect...
Device 'wlan0' successfully activated with '43c8b23a-758e-43e8-9c8a-53bc16639e32'.
Wiping NetworkManager configs
...
Found access point name: "Merkury-5D5E", trying to connect...
Device 'wlan0' successfully activated with 'be5d9eda-88a4-4109-a7b2-a9b56c01679d'.
Wiping NetworkManager configs
..
Found access point name: "Merkury-5D5E", trying to connect...
Device 'wlan0' successfully activated with 'd33cb2de-8f40-4082-b7e4-12f20b270e52'.
Error, could not connect to SSID.
Failed to connect, please run this script again`

Cossid, 2024-05-05:
1) Make sure the devices are nearby enough to each other
2) Try running with the `-r` parameter if you're not already

This is an issue with NetworkManager not able to fully connect it appears

bigboigahoy, 2024-05-05:
> 1. Make sure the devices are nearby enough to each other
> 
>     2. Try running with the `-r` parameter if you're not already
> 
> 
> This is an issue with NetworkManager not able to fully connect it appears

I moved the devices closer and I still have the same issue. I was able to get some of the bulbs working correctly, it seems the MI-BW904 is giving me problems specifically. I was using the -r parameter as well.

akintash, 2024-06-05:
The same issue with Hombli_HBSS-0209-Smart-Socket-B2030248-Energy-Plug.
FW version 1.0.2.
Tried different WLAN adapters (Broadcom and Realtek), issue happens on both of them.

bigboigahoy, 2024-06-11:
> The same issue with Hombli_HBSS-0209-Smart-Socket-B2030248-Energy-Plug. FW version 1.0.2. Tried different WLAN adapters (Broadcom and Realtek), issue happens on both of them.

I wonder if there is a firmware bug that effects these. I have had success with numerous other devices as well as the same device, just not this specific firmware.

beasthouse-au, 2024-08-09:
> > The same issue with Hombli_HBSS-0209-Smart-Socket-B2030248-Energy-Plug. FW version 1.0.2. Tried different WLAN adapters (Broadcom and Realtek), issue happens on both of them.
> 
> I wonder if there is a firmware bug that effects these. I have had success with numerous other devices as well as the same device, just not this specific firmware.
> 
> Edit: I did get a different error: `Found access point name: "Merkury-B789", trying to connect... Error: Connection activation failed: (11) 802.1X supplicant took too long to authenticate.`

I am getting this same error. Other devices of the same model worked, this one does not. Arlec PC191HA Smart Plug

metaljay, 2024-11-14:
Sorry to bump and old thread - I also get this error just with a generic starlight projector, running on a raspberry pi (followed the instruction):
```
[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231t_light3_laser_nanxin
 ► 1.1.2 - BK7231T / oem_bk7231t_light3_laser_nanxin

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 ► ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
   esphome_2024.10.3_generic-bk7231t-qfn32-tuya_bk7231t_lt1.7.0.uf2
   image_bk7231t_app.ota.ug.bin
   OpenBeken-v1.17.697_bk7231t.ug.bin

Selected Device Slug: tuya-generic-sk20-star-projector-v1.1.2
Selected Profile: oem-bk7231t-light3-laser-nanxin-1.1.2-sdk-1.0.2-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
......
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Wiping NetworkManager configs
..
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Wiping NetworkManager configs
..
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Wiping NetworkManager configs
..
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Wiping NetworkManager configs
..
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Error, could not connect to SSID.
Failed to connect, please run this script again
```

Cossid, 2024-11-14:
@metaljay That is purely a connection issue, the device you're running CloudCutter from does not have a good enough wireless connection to the device you're trying to exploit.

metaljay, 2024-11-14:
> @metaljay That is purely a connection issue, the device you're running CloudCutter from does not have a good enough wireless connection to the device you're trying to exploit.

they are next to one another, as in. my raspberry pi (connected via ethernet) is sat on top of the star projector. I suspect it's probably something in the pi settings that's stopping it communicating properly over wifi? I dont have any other linux based devices to try this on, my Mac can't virtualise Ubuntu and have a wifi connection, it just shows as ethernet :(

Cossid, 2024-11-14:
Not much else I can advise,

```
Found access point name: "SmartLife-BB3D", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
```

is purely a driver-level response, that error isn't CloudCutter code.

closed 2025-01-09 by Cossid