tuya-cloudcutter/tuya-cloudcutter #663: Tuya LED BULB W5K - bluetooth and wifi.

flomabe, 2024-05-14:
Hello cloudcutters,

I'm very new to this - recently bought a cheap Tuya bulb from AliExpress:
https://www.aliexpress.com/item/1005006253955848.html

It was just around 3 bucks (with choice) and is now even closer to 2 - but before I buy more I'd like to make sure I can actually free them from the cloud.
That's a steal compared to those pre-flashed bulbs by Athom.

So far my attempts using tuya-cloudcutter all failed and, no matter what profile I selected, ended with 

`[!] The profile you selected did not result in a successful exploit.`

However, when I "reset" the device (toggle on-off 5 times) the bulb always blinks fast (red light) - I'm not able to put it in slow blink mode - but the AP is opened anyway and tuya-cloudcutter is able to connect.

According to the tuya smart app on Android the name of the device is "LED BULB W5K" (given by the app before editing/renaming) with
 
Main Module: V1.0.0
MCU Module: V1.0.0

Unfortunately I have no other device to provide a firmware dump with lightleak.

Any hints if this device is supported somehow would be appreciated.

Maybe should I try to get a wifi only version? Worth a shot?

Thanks!

Cossid, 2024-05-14:
Not a lot of advice we can give, 1.0.0 is a generic version number, and often belongs to device-specific firmwares.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out is where I'd start to narrow down what might be inside.  Otherwise you may need to open to verify.

The ali express product link you provided does not work for me in the US.

flomabe, 2024-05-15:
> https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out is where I'd start to narrow down what might be inside. Otherwise you may need to open to verify.

The MAC address starts with B8:06:0D, Tuya Smart Inc, so it's not a Espressif, if Beken or Realtek I don't know - guess I have to open it then to find out.

>The ali express product link you provided does not work for me in the US.

Yes, same happens to me if I select the US as a country to ship to.
The shop/seller selling these bulbs seems to ship to Europe exclusively. If logged in and selecting a EU country or the UK it should show the article - but I guess this isn't of much help anyway.
So far I couldn't find a seller that offers those dual BT/Wifi bulbs shipping to the US, but I'll keep looking. 

So, to get lightleak running, I'd need any device with ESP32/ESP8266/BK7231 or RTL8710B with LibreTiny? 
Would those athom bulbs pre-flashed with ESPHome fit the bill?

Cossid, 2024-05-15:
You generally don't want to use a regular device for the lightleak portion, you'd want a dev board.

Lightleak is a bit difficult, and not a guarantee, especially if you haven't verified exactly which chip/module is inside your bulb.

flomabe, 2024-05-18:
So, I opened the bulb and the chip used is the Lightning LN882H(KI) - seems I'd need to flash it manually.

https://www.elektroda.com/rtvforum/topic4028087.html
https://www.elektroda.com/rtvforum/topic4037269.html

Unfortunately when I opened it the 5th time I ripped the 2x3 pin header off the LED board, which was fixated with a crazy amount of silicone (which I peeled of earlier to get it open in the first place).
Maybe I can solder it back on.

I ordered 4 different wifi only bulbs from AE and hope I'll have more luck with them.

So I'd say this issue can be closed. Thanks a lot anyway - I learned a ton so far.

closed 2024-05-18 by Cossid