tuya-cloudcutter/tuya-cloudcutter #664: Issue when trying to build docker image

kennyyy53, 2024-05-14:
I tried a lot of time to execute the command : sudo docker build --network=host -t cloudcutter . 
Each time, it seams like my raspberry disconnect himself of internet when he is creating virtual environnement with python.
[image]

Cossid, 2024-05-14:
That's not really cloudcutter, that's docker, there is no custom code there.  It should be changing any network settings, so there is no reason for a disconnect.

Seems like you have general issues that we really can't support.

kennyyy53, 2024-05-15:
Yes sry, I know that this is not really cloudcutter but this is a problem that I have when I am doing the procudure of installation.

I am not sure that this is a connection issue but the process stop every at the point that we saw on the screen

closed 2025-01-09 by Cossid