tuya-cloudcutter/tuya-cloudcutter #665: PSKKEY/AUTHKEY do not match device.

jirkag, 2024-05-16:
Hi,
I have LSC 3000272 E27 806lm bulb. I tried to flash it with 3000272 choice in LSC menu. It did not work. I have checked firmware and found that firmware is 1.3.1 version. Cutting worked fine, but I got "PSKKEY/AUTHKEY do not match device" error. 
Please can you help me?
Verbosed log:

Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.3.1 - BK7231N / oem_bk7231n_light_ty
   1.2.21 - BK7231N / oem_bk7231n_light_ty
   1.2.6 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty
   1.2.8 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.2.8 - BK7231T / oem_bk7231_2onoff_wetemp
   1.3.0 - BK7231N / kltwfb_single_winctrl_rf_lns00
 > 1.3.1 - BK7231N / oem_bk7231n_light_ty
   1.3.19 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.3.2 - BK7231T / oem_bk7231s_dltj
   1.3.2 - BK7231T / oem_bk7231s_light_pir_globe
   1.3.21 - BK7231N / oem_bk7231n_light_ty
   1.3.22 - BK7231N / oem_bk7231n_light_ty

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: moes-14w-rgbw-bulb
Selected Profile: oem-bk7231n-light-ty-1.3.1-sdk-ffcgroup-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
..................
Found access point name: "A-FC66", trying to connect...
Device 'wlan0' successfully activated with '5052757b-7f43-499c-bd4c-8e081ecbdde0'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/zSQNwuqAUwv0.deviceconfig
Saved device config in /work/configured-devices/zSQNwuqAUwv0.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
[... 84 lines cut ...]
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 DHCPACK(wlan0) 10.42.42.17 fc:67:1f:ae:fc:66 wlan0
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  5
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
May 16 14:05:39 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
May 16 14:05:57 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.17
May 16 14:05:57 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01

[14:05:58.602204 Log (Client)] Request: HTTPServerRequest(protocol='https', host='h3.iot-dns.com', method='POST', uri='/v2/url_config', version='HTTP/1.1', remote_ip='10.42.42.17')
[14:05:58.602515 LOG (Client)] ==== Request body ===
{"region":"AA","env":"","uuid":"BAdE2GtCyg3X","ts":53,"sign":"gu7gxWJbv4zTM0CHx7hQUzF1vmYRN0QSjPMt2RRQhC0=","config":[{"key":"httpUrl","need_ca":false},{"key":"mqttUrl","need_ca":false},{"key":"httpsPSKUrl","need_ca":false},{"key":"mqttsPSKUrl","need_ca":false},{"key":"ghostUrl","need_ca":true}]}
[14:05:58.602677 LOG (Client)] ==== End request body ===
[14:05:58.603052 LOG (Server)] Response:  {"caArr":null,"httpUrl":{"addr":"http://10.42.42.1/d.json","ips":["10.42.42.1"]},"mqttUrl":{"addr":"10.42.42.1:1883","ips":["10.42.42.1"]},"ttl":600}
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
[!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.
Oh no, something went wrong with updating firmware! Try again I guess...

Cossid, 2024-05-16:
This means you successfully exploited the device with a correct profile to give it the `A-` prefix, and then used a different incorrect profile late.  You'll need to identify the original one you used that worked.

jirkag, 2024-05-16:
Found it, sorry for waisting your time

closed 2024-05-16 by jirkag