tuya-cloudcutter/tuya-cloudcutter #666: new device Calex smart multi color LED strip UPK2Home dump

spanner3003, 2024-05-17:
I'm not sure where to put these file that a pulled off the Calex smart multi color LED strip I brought from tesco years ago but i used esphome-kickstarter and UPK2ESPHome grab for kickstarter option hope they are helpful.
[Merkury EW003 LED Strip - schema ID 000002hqnb.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15355599/Merkury.EW003.LED.Strip.-.schema.ID.000002hqnb.json)
[Merkury EW003 Smart LED Strip Storage Data.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15355600/Merkury.EW003.Smart.LED.Strip.Storage.Data.json)
This is the box they came in:
[image]

_Originally posted by @spanner3003 in https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/33#issuecomment-2118171138_

Cossid, 2024-05-18:
Your file names say Merkury Innovations EW003 (and seem to support that with the data inside) , and the title/image is for Calex.  Are you saying the Merkury product came in a Calex box?

spanner3003, 2024-05-18:
yes I am it looks like calex Holland is pass Merkury of as its own.

Cossid, 2024-05-21:
Added as `Merkury Innovations MI-EW003 LED Strip v1.1.6` as that is what it would show up under SmartLife as.

closed 2024-05-21 by Cossid