tuya-cloudcutter/tuya-cloudcutter #668: Tuya Human Presence Detector ZG-205-W doesn't goes to AP mode

LitBull, 2024-05-20:
I've tried to convert a Tuya Human Presence sensor **ZG-205-W** (Wifi) but it is imposible set it in **AP** mode.
I've tried shutting up and down 6 times and then 3 more times without success. 
I tried to do it by the Smarlife App but it can't be set into manual mode as it doesn't appears its reference, so it can't be set into AP mode from the app.
The only way to add a device to the Smartlife app is in automatic mode.
The model is a ZG-205-W, Main Module: V2.1.6 , MCU Module V1.0.5

bigboigahoy, 2024-05-28:
I have a ir blaster with temperature and humidity display that also has this issue. I see that other versions that seem to work, but for some reason this model seems to lack it. [](https://www.aliexpress.us/item/3256806253507221.html?spm=a2g0o.order_list.order_list_main.18.75341802Vj0Gpk&gatewayAdapt=glo2usa)

Cossid, 2025-01-09:
I believe it has been verified that this specific device does not have/support an AP mode, which means CloudCutter cannot run on this device, despite us having been able to generate a profile from a serial firmware dump.

closed 2025-01-09 by Cossid