tuya-cloudcutter/tuya-cloudcutter #669: LSC Smart Led Strip 3203632

Xenuzz, 2024-05-21:
Hello, I recently bought an LSC Smart Led Strip: https://www.action.com/de-at/p/3203632/lsc-smart-connect-led-stange/
Firmware: Main module: V1.1.2
MCU module: V1.1.2
I wanted to ask if you could create a profile for me. Thanks in advance

Cossid, 2024-05-21:
We can't create a profile without a firmware dump.  If you are able to provide one, feel free to attach it here and re-open this issue.

closed 2024-05-21 by Cossid

Xenuzz, 2024-05-24:
Okay, I have the same device again now and it has firmware 1.1.2 on it and I could send you a copy of the firmware tomorrow. Thx

divadiow, 2024-12-04:
fw v1.0.2 from an LSC 3203632.1

https://www.elektroda.com/rtvforum/topic4091111.html
[readResult_BK7231N_QIO_LSC-5M_2024-01-12-17-10-09.zip](https://github.com/user-attachments/files/18007532/readResult_BK7231N_QIO_LSC-5M_2024-01-12-17-10-09.zip)
[boot.txt](https://github.com/user-attachments/files/18007581/boot.txt)

Cossid, 2024-12-04:
So this one is an oddball, we actually did receive a dump for firmware 1.1.2, and it was patched
1.1.2 was used SDK 3.3.40, and this 1.0.2 firmware uses a much newer SDK of 3.8.5

To clarify, both 1.1.2 and 1.0.2 are both patched in terms of CloudCutter support.