tuya-cloudcutter/tuya-cloudcutter #670: tuya-generic-universal-ir-remote-control-cb3s-v2.0.0

kronicnz, 2024-05-22:
E: UPK: No actual components were added!
This means that **valid configuration has been found**,
but this type of device is not yet supported by this program.
This includes for example thermometers, water leak sensors, or fan controllers.
E: An error has occurred
E: RuntimeError: Error: FIRMWARE_NOT_MATCH: Non-matched firmware
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltctplugin\upk2esphome\work_schema.py", line 62, in run_impl

```
{
    "manufacturer": "Tuya Generic",
    "name": "Universal IR Remote Control CB3S v2.0.0",
    "key": "key54vrth5askhsj",
    "ap_ssid": "SmartLife",
    "github_issues": [],
    "image_urls": [
        "https://tuya-cloudcutter.github.io/images/tuya-generic-universal-ir-remote-control-cb3s-v2.0.0.jpg"
    ],
    "profiles": [
        {
            "name": "2.0.0 - BK7231N",
            "sub_name": "oem_bk7231n_irbox_mol_ty",
            "type": "CLASSIC",
            "icon": "memory",
            "firmware": {
                "chip": "BK7231N",
                "name": "oem_bk7231n_irbox_mol_ty",
                "version": "2.0.0",
                "sdk": "2.3.1-40.00",
                "key": "key54vrth5askhsj"
            },
            "data": {
                "address_finish": "0x1B5B7",
                "address_ssid": "0xB40C5",
                "address_ssid_padding": 4
            },
            "slug": "oem-bk7231n-irbox-mol-ty-2.0.0-sdk-2.3.1-40.00",
            "devices": [
                {
                    "slug": "aubess-ir-remote-controller",
                    "manufacturer": "Aubess",
                    "name": "IR Remote Controller",
                    "image_url": "https://tuya-cloudcutter.github.io/images/thumbs/aubess-ir-remote-controller.jpg"
                },
                {
                    "slug": "tuya-generic-s18-ir-repeater",
                    "manufacturer": "Tuya Generic",
                    "name": "S18 IR Repeater",
                    "image_url": "https://tuya-cloudcutter.github.io/images/thumbs/tuya-generic-s18-ir-repeater.jpg"
                },
                {
                    "slug": "tuya-generic-universal-ir-remote-control-cb3s-v2.0.0",
                    "manufacturer": "Tuya Generic",
                    "name": "Universal IR Remote Control CB3S v2.0.0",
                    "image_url": "https://tuya-cloudcutter.github.io/images/thumbs/tuya-generic-universal-ir-remote-control-cb3s-v2.0.0.jpg"
                },
                {
                    "slug": "tuya-generic-universal-ir-remote-control-cbu-v2.0.0",
                    "manufacturer": "Tuya Generic",
                    "name": "Universal IR Remote Control CBU v2.0.0",
                    "image_url": "https://tuya-cloudcutter.github.io/images/thumbs/tuya-generic-universal-ir-remote-control-cbu-v2.0.0.jpg"
                }
            ],
            "github_issues": [
                166,
                249
            ]
        }
    ],
    "schemas": {
        "000003t4o3": [
            {
                "mode": "rw",
                "property": {
                    "type": "string",
                    "maxlen": 3072
                },
                "id": 201,
                "type": "obj"
            },
            {
                "mode": "ro",
                "id": 202,
                "type": "raw"
            }
        ]
    },
    "device_configuration": {
        "crc": 78,
        "infre": 26,
        "infrr": 24,
        "netnc": 0,
        "netyc": 1,
        "owm": 0,
        "reset_lv": 0,
        "reset_pin": 6,
        "rsthold": 3,
        "wfst_lv": 1,
        "wfst_pin": 7
    },
    "slug": "tuya-generic-universal-ir-remote-control-cb3s-v2.0.0",
    "image_url": "https://tuya-cloudcutter.github.io/images/thumbs/tuya-generic-universal-ir-remote-control-cb3s-v2.0.0.jpg"
}
```

Cossid, 2024-05-22:
There is nothing actionable here for CloudCutter.  This isn't the support for UPK, and UPK doesn't support all device classes, it is only meant as a tool for common devices and to basic information about the device to get you started on a more comprehensive 3rd party firmware config.

closed 2024-05-22 by Cossid

kronicnz, 2024-05-23:
Thanks.. So that deems my device as useless now?

Cossid, 2024-05-23:
No, you just need to figure out how to configure it if you installed a 3rd party firmware.  IR TX is on P26, IR RX is on P24, the button is on P6, and the LED is on P7.  The error you posted just means that UPK doesn't know how to make a config for that type of device (and is unrelated to CloudCutter).

kronicnz, 2024-05-23:
Thanks for your replies.. I am new to this cloudcutter concept, not really understanding the fix for this problem :( can you point me to some content that will help me understand better where/how to manually configure this device? Thanks!

Cossid, 2024-05-23:
Nothing is broken, just unsupported (and again, not CloudCutter, you're past anything CloudCutter can do).  You're at the point where you need to configure 3rd party firmware.  You can look at existing examples for ESPHome at https://devices.esphome.io/type/misc and search for IR devices to use as an example.