tuya-cloudcutter/tuya-cloudcutter #673: Amico Smart Ceiling Fan

sredman, 2024-05-26:
I have a Tuya-powered smart ceiling fan: https://www.amazon.com/dp/B0BWJQGQ7K

It is running firmware v2.1.17, so known not vulnerable, however the serial pins are easily access so it's easy to take a dump.

The build-profile script fails saying it is patched. If I comment out that step, it next fails and says `[!] Failed to find ssid payload address (found 0, expected 1)`

Even though the firmware is patched, I was hoping to get the profile .json to use to make an ESPHome configuration. Just getting the pins working isn't too hard, it seems to be just 3x PWM ins, but I don't know how to make the IR remote work, and I'd really like to keep that. Uploading the storage .json to [UPK2ESPHome](https://upk.libretiny.eu/) fails, saying "No schema model for TuyaMCU - can't process datapoints".

Anyways, since it was easy to dump I thought it would be nice to share it here. I need to mount this ceiling fan ASAP, so I probably won't be able to test anything, but maybe the next traveler will find it helpful.

Attached .zip contains two .bin dumps and the corresponding storage.json. One .bin was taken with the Tuya app paired, the other .bin was after wiping.

[Amico Smart Ceiling Fan.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15448652/Amico.Smart.Ceiling.Fan.zip)

sredman, 2024-05-26:
This device shows up in the Tuya app as just "Smart Ceiling Fan", with no custom device icon, so it seems like it might be a generic controller. In that spirit, here are pictures of the board and the case to help others potentially identify if they're holding the same device. The wires on the bottom of the board are of course not original, and were soldered by me. I've labeled my connectors in case someone wants to follow in my footsteps

Exterior:
[image]
[image]
[image]

Interior:
[image]
[image]
[image]
[image]

Cossid, 2024-05-26:
Yeah, nothing further on the CloudCutter front, the device is patched.  Looks like a TuyaMCU device, where the fan is controlled through the TuyaMCU uart protocol on RX1/TX1 at baud rate 9600.

DPIDs that would need to be mapped out:

```json
"000004tqqq": [{
        "type": "obj",
        "mode": "rw",
        "property": {
            "type": "bool"
        },
        "id": 20
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "min": 10,
            "max": 100,
            "scale": 0,
            "step": 1,
            "type": "value"
        },
        "id": 22
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "min": 1,
            "max": 3,
            "scale": 0,
            "step": 1,
            "type": "value"
        },
        "id": 23
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "type": "bool"
        },
        "id": 60
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "range": [
                "fresh",
                "nature"
            ],
            "type": "enum"
        },
        "id": 61
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "min": 1,
            "max": 6,
            "scale": 0,
            "step": 1,
            "type": "value"
        },
        "id": 62
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "range": [
                "forward",
                "reverse"
            ],
            "type": "enum"
        },
        "id": 63
    }, {
        "type": "obj",
        "mode": "rw",
        "property": {
            "range": [
                "cancel",
                "1h",
                "2h",
                "4h",
                "8h"
            ],
            "type": "enum"
        },
        "id": 101
    }
]
```

closed 2024-05-26 by Cossid