tuya-cloudcutter/tuya-cloudcutter #678: After the cut device is read only

krishgcek, 2024-06-11:
After I cut the smart plug, with tiny tuya I can read the status but turning on the dp 1 is not working.

Cossid, 2024-06-11:
What device is this?  What profile did you use to cut it?

krishgcek, 2024-06-11:
I have used 16 amps  generic device .main module 1.1.8.

It is a Wipro 16 amps smart plug.
Sdk , oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

Cossid, 2024-06-11:
But _which_ of the `oem-bk7231n-plug-1.1.8-sdk-231-40.00` device profiles did you choose?  The each one has potentially different dpids.

Cossid, 2024-06-11:
I guess it might not matter, all 28 profiles have DPID 1 available.

If you can read the status, the device is working, you're likely doing something incorrect with Tiny Tuya.  Additionally you can try cutting again with the same or a different profile for the same firmware version.

krishgcek, 2024-06-11:
Can I use sudo ./tuya-cloudcutter.sh -p oem-bk7231n-plug-1.1.8-sdk-231-40.00.
If I put -v I see some dns waiting messages

Cossid, 2024-06-11:
Using `-p` will automatically choose the first profile available it can find, but it will also list out the slug of the one it chose in the output.

krishgcek, 2024-06-11:
> I guess it might not matter, all 28 profiles have DPID 1 available.
> 
> If you can read the status, the device is working, you're likely doing something incorrect with Tiny Tuya. Additionally you can try cutting again with the same or a different profile for the same firmware version.

Here is the tinytuya debug,
DEBUG:sending payload
DEBUG:payload encrypted=b'000055aa000000010000000700000077332e33000000000000000000000000257bc49ed11cc58cf5001a620d5b1990bb0392da6b4c40f1b6a8bbf34c5965a83284757646c4b06ef46029e40ce39f655c15a6a3824b4782cb6525b817c8dd22c9913d784086cdcd5c7de4e69dd104dd72ed877f91851fe767044ef99cd82611e38106ac0000aa55'
DEBUG:received data=b'000055aa00000001000000070000000c00000000a505a9140000aa55'
DEBUG:received null payload (TuyaMessage(seqno=1, cmd=7, retcode=0, payload=b'', crc=2768611604, crc_good=True, prefix=21930, iv=None)), fetch new one - retry 0 / 5
DEBUG:set_status received data=None

Cossid, 2024-06-11:
You'll need to reach out to Tiny Tuya for support.  Every profile CloudCutter can send to the device supports read and write for DPID 1.

closed 2024-06-11 by Cossid