tuya-cloudcutter/tuya-cloudcutter #684: QS-WIFI-S10-C04 Curtain module available on CBLC9

MichalTJanas, 2024-07-02:
Hello, is there any chance to add to the support of this Curtain Module uses the [CBLC9](https://developer.tuya.com/en/docs/iot/cblc9-module-datasheet?id=Ka42cqnj9r0i5)/BK7231N

Are found on boards with the designation: Board ID: QL-0887, Rev. A, 2024.01.23
or
Board ID: AH-A163, Rev. C, 2024.05.25

Thanks in advance.

closed 2024-07-02 by Cossid

MichalTJanas, 2024-07-02:
There is his [image]

Cossid, 2024-07-02:
There is not currently a profile for 2.0.1 curtain switches, a firmware dump would be needed to support the device, assuming it is not also already patched.

MichalTJanas, 2024-07-02:
I will try to solder CBLC9 and rip the firmware as soon as I find a moment.

MichalTJanas, 2024-07-03:
Hi,
I attach a dump from QS-WIFI-S10-C04 and output:

```
{
	"base64":"AgBapQAAGwAAYOoAAA=="
}

Sorry, no meaningful pins data found. This device may be TuyaMCU or a custom one with no Tuya config data.
No module information found.
And the Tuya section starts, as usual, at 2023424
```

[QS-WIFI-S10-C04-dump.zip](https://github.com/user-attachments/files/16086945/QS-WIFI-S10-C04-dump.zip)

Cossid, 2024-07-03:
Added to CloudCutter as `Tuya Generic QS-WIFI-S10-C04 Curtain Module v2.0.1`

This doesn't have any GPIO configured, so may indeed mostly be TuyaMCU, in which case I'm not certain how that maps for a curtain device.

MichalTJanas, 2024-07-03:
I found on the Elektroda forum as below:
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "QS-WIFI-S10-C04",
  "model": "Mini WiFi Curtain Module",
  "chip": "BK7231N",
  "board": "CBLC9",
  "flags": "1024",
  "keywords": [
   "shutters",
    "blinds",
    "TODO"
  ],
  "pins": {
    "6": "Btn;1",
    "7": "Rel;2",
    "8": "WifiLED_n;0",
    "9": "Rel;1",
    "24": "Btn;2",
    "26": "Btn;1"
  },
  "command": "PowerSave 1",
  "image": "https://obrazki.elektroda.pl/1978468300_1713373135.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/viewtopic.php?p=21049246"
}
```

Cossid, 2024-07-03:
Ah.  Yeah, in re-looking, it appears this is a firmware made specifically for this device, so if it has gpio, it was programmed into the firmware instead of storage configuration, so we can't extract it, and it looks the same as an MCU device (empty config).