tuya-cloudcutter/tuya-cloudcutter #685: Curtain module QS-WIFI-C01

skowron0, 2024-07-03:
Hi,
I have full read ROM from curtain module QS-WIFI-C01.  Inside is installed LMB54 module with BK7231N. I have over 10 such modules installed and I would like to upload esphome to them without disassembling them. Could someone add a profile for this module?

[qs-wifi-c01.zip](https://github.com/user-attachments/files/16078164/qs-wifi-c01.zip)

Cossid, 2024-07-03:
Added as `Tuya Generic QS-WIFI-C01 Curtain Module v1.0.5`

Note, you mentioned yesterday version 2.0.1, but this is 1.0.5.  You were either looking at the wrong device, or have multiple firmware versions.  We'd need a dump for each firmware version.  There was no upgrade available for the 1.0.5 firmware.

closed 2024-07-03 by Cossid

skowron0, 2024-07-04:
At start, I Thank You for adding my device.
I have problem when I running script. Always stop on:
```
================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr d8:3a:dd:f3:47:04 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```
I wait one hour on this step and notting happened

this is all log:

[log.txt](https://github.com/user-attachments/files/16094572/log.txt)