tuya-cloudcutter/tuya-cloudcutter #689: Device Dump - NOVOSTELLA Permanent Outdoor Lights, Smart RainbowColor RGB IC Light, 100ft (50x2) with 80 LED

ak-ha-guy, 2024-07-16:
‎NVS100-RGBIC-US-F
[image]

[ltchiptool_bk72xx_2024-07-14_22-23-11.zip](https://github.com/user-attachments/files/16243586/ltchiptool_bk72xx_2024-07-14_22-23-11.zip)

Cossid, 2024-07-16:
The app partition appears to have some abnormalities.  Did you use other tools prior to ltchiptool?  Could you try dumping again?

ak-ha-guy, 2024-07-16:
I already flashed it to ESPHOME Kickstarter.   I did back up a second one before flashing it as well. 
[Uploading ltchiptool_bk72xx_2024-07-14_22-17-02-01.zip…]()

Cossid, 2024-07-16:
That latest attachment did not come through correctly.

ak-ha-guy, 2024-07-16:
[ltchiptool_bk72xx_2024-07-14_22-17-02-01.zip](https://github.com/user-attachments/files/16254579/ltchiptool_bk72xx_2024-07-14_22-17-02-01.zip)

ak-ha-guy, 2024-07-16:
Error I get when trying to open either. 
[image]

Cossid, 2024-07-16:
That one has the same issue, it has CRC errors, and doesn't look like the app partition would boot.

We can brute-force decrypt it (ignoring the CRC errors), and it is using an unusual `0.0.1` SDK.  Because the app partition is busted, I can't really generate a profile for this one reliably.

Because it is RGBIC, UPK wouldn't generate a profile for it anyway.  It would likely be configured as SPI, https://esphome.io/components/light/beken_spi_led_strip.html on ESPHome.

ak-ha-guy, 2024-07-16:
I think you are correct.  The problem is I can not figure out what LED driver chip it is.  
WS2812
SK6812
APA106
SM16703

It's fully encased with no markings. 
[image]
[image]

ak-ha-guy, 2024-07-16:
That is P16 going to the LEDs

Cossid, 2024-07-16:
Most Likely WS2812, there are WS2812B string references in the part of the app that does extract.  SM16703 is also WS2812 protocol from what I understand.

ak-ha-guy, 2024-07-16:
Thanks.  I tried all the ones listed above without luck.  I tried WS2812B and it gave an error.  Maybe WS2812B is not in the platform or bk72xx: board: generic-bk7231n-qfn32-tuya. ?

`esphome:
  name: "led-house-outside-west"
  friendly_name: House West RGB LEDS

bk72xx:
 board: generic-bk7231n-qfn32-tuya

# serial connection to Tuya MCU
uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600
# Enable logging

logger:

# Enable Home Assistant API
api:
  encryption:
    key: "......"

ota:
  - platform: esphome
    password: "......"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kickstart-Bk7231N"
    password: ">>>>>"

captive_portal:

light:
  - platform: beken_spi_led_strip
    rgb_order: GBR
    pin: P16
    num_leds: 40
    id: rgb_led
    chipset: WS2812
    is_wrgb: true
    name:  "Light"
`

Cossid, 2025-06-10:
This has veered outside the realm of CloudCutter support, as without a valid dump, there is no new profile that can be added.  Support for 3rd party firmware configuration should be sought at their respective places for support.

closed 2025-06-10 by Cossid