tuya-cloudcutter/tuya-cloudcutter #690: Profile update needed For two different feit electric 2019 smart bulbs.

rubin9family, 2024-07-17:
Profile update for 2 different 2019 Feit smart bulbs:
OM60/RGBW/CA/AG (P) with firmware version v1.0.7. Revision is 228-19-46.
BPA800/RGBW/AG/2 (P) with firmware version v1.1.5. Revision is 228-19-38.

Also the instructions about how to make your own profiles for cloudcutter needs to be explained better. I wasn't able to follow nor was i able to figure out how to make a dump of my firmware.

Cossid, 2024-07-17:
Creating new profiles explicitly requires a dump, as that is the only way to get the data.

Dump instructions are in the core readme of the project.

rubin9family, 2024-07-17:
i need help getting dump. what program can i use to wirelessly get the dump from a tuya lightbulb?

Cossid, 2024-07-17:
The only way to get wireless dumps is via LightLeak which requires and android phone, sideloading an app, and a serial programmed dev board, and even then results are not guaranteed, it is not a very stable process, and is not very user friendly.

rubin9family, 2024-07-17:
so how do i do it wired with having to buy to many things?

Cossid, 2024-07-17:
You would need a USB TTL Uart device, some wire, and ability to disassemble & solder (or a BDM frame/probes)

However, you are not guaranteed to get a profile if any of the devices have patched firmware.

I'm scanning based on assumptions to see if I can find upgrades to these firmware versions, however it will take both time and luck, you're welcome to wait on the results of that.

rubin9family, 2024-07-17:
Thanks! I'll wait. The light bulbs are definitely not patched.  the 2022 version I have on their latest firmware isn't patched so certainly the 2019 version with an older software version shouldn't be patched.

Cossid, 2024-07-17:
2019 could also be ESP8266, which wouldn't be compatible with Tuya-CloudCutter (but possibly Tuya-Convert).
Is there an FCC ID listed on these bulbs?

Alternately, you can partially verify with https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

rubin9family, 2024-07-17:
they are ESP82 series. Thank you for helping me figure that out. I'll try Tuya convert. If you have any tips about Tuya convert, please let me know.

Cossid, 2024-07-17:
Not much.  The only thing I can say is sometimes the OS matters, but I'm not sure why.  Some devices work on RPI Bullseye and newer, some only work on Buster, and it's pretty random.

closed 2024-07-17 by Cossid

rubin9family, 2024-07-17:
Do I have to use a Raspberry Pi for Tuya convert or a Linux laptop with an ethernet port?

Cossid, 2024-07-17:
You need a wireless card, but RPI isn't a requirement.  Bullseye/Buster codenames actually apply to the underlying Debian distro.

szupi-ipuzs, 2024-07-17:
> they are ESP82 series. Thank you for helping me figure that out. I'll try Tuya convert. If you have any tips about Tuya convert, please let me know.

I've recently used tuya-convert on 2 devices succesfully (and unsuccesfully on other 2), and here are my tips:
1) You don't actually need a separate PC or RPi. The alternative is: a usb wifi dongle + virtualbox installed + a [freely available ubuntu image](https://ubuntu.com/tutorials/how-to-run-ubuntu-desktop-on-a-virtual-machine-using-virtualbox#1-overview). When the virtual ubuntu runs, you can pass it the control over the usb dongle.
2) Use the [development branch](https://github.com/ct-Open-Source/tuya-convert/tree/development)!
3) Remember to set the proper name of the network interface in the tuya-convert .ini file.
4) If smart-pairing mode doesn't work, try with AP mode and a script from [this PR](https://github.com/ct-Open-Source/tuya-convert/pull/966).
5) Be patient, do many retries :)