tuya-cloudcutter/tuya-cloudcutter #695: TAXNELE-TVPS1-63T

flammable47, 2024-08-08:
Hello
I would like to ask of this model is supported on Tuya-cloudcutter?
[image]
[image]
[image]
[image]

Here the like of where I brought it
https://shopee.ph/product/277704323/20954963907?gads_t_sig=VTJGc2RHVmtYMTlxTFVSVVRrdENkVlBXTnFLbGtLY21IOVhMT0xMVGhrdnNRSFJIZERRTEN0dHlQUEtHb3FadEhOaEdjZ0VIK1BBeGgxUzc3M3VUOERzeEZUNTNRN3RMVjNvZ20xOEFRSldwY2pJTWRnQThteHpNTzlIKzRKaG8&gad_source=1&gclid=EAIaIQobChMI8fjXjNfkhwMVX1YPAh0GrzltEAQYAyABEgLshvD_BwE

flammable47, 2024-08-08:
[ltchiptool_bk72xx_TAXNELE-TVPS1-63T-Dump-File.zip](https://github.com/user-attachments/files/16538178/ltchiptool_bk72xx_TAXNELE-TVPS1-63T-Dump-File.zip)

Cossid, 2024-08-08:
Added as `TAXNELE TVPS1-63LW Circuit Breaker v2.1.6`

It's a TuyaMCU device at baud 9600.  Actual 3rd party firmware support status is unknown.

closed 2024-08-08 by Cossid