tuya-cloudcutter/tuya-cloudcutter #698: Using Tuya cloud cutter for 1.3.16 firmware version?

XenisIII, 2024-08-20:
Hello,
Recently, i have bought aubess smart plug (https://fr.aliexpress.com/item/1005005601443212.html?spm=a2g0o.order_list.order_list_main.16.19165e5bawYmbb&gatewayAdapt=glo2fra). I had hope to flash openBK firmware on micro-controller because i had seen this plug on repertoried devices. But the installed firmware version is 1.3.16 and i don't see it in the firmware version on cloud cutter.

I opened the plug and surprise! I have found T34 2133k9t4j module. I have found a tutoriel to flash this module with UART method (https://www.elektroda.com/rtvforum/topic4042412.html) but it doesn't work for me. The ToolFlasher can't connect to the bus.

I would like to know if there are plans to integrate this version with tuya cloud cutter? (I tried everything with the UART method but nothing worked... I'm desperate. My only hope is tuya cloud cutter)

Thank you and have nice day ^^

virtorgan, 2024-08-31:
Same situation here, I got a device T34 / BL0937 with 1.3.16 firmware.

Same as here :[https://www.elektroda.com/rtvforum/topic4042412.html](url)
Would like to have this integrated.
Even if the job already done is amazing.

Cossid, 2024-09-03:
1.3.16 is newer than 1.3.5/1.3.8/1.3.10 which are known patched, and would also be assumed as patched.

closed 2024-09-03 by Cossid