tuya-cloudcutter/tuya-cloudcutter #699: Need update for Z2-N-V1.1 module running v1.0.4

dev462, 2024-08-22:
I have a WP9 power strip that has a Z2-N-V1.1 in it that reports as having both Main and MCU running v1.0.4

reading through https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/598 they appear to be the exact same module except with different versions loaded. Is it possible to get an OTA update to include v1.0.4 for the Z2-N-V1.1 or is there more information needed from the device? Please let me know an I'll pull down what ever i can.

Cossid, 2024-09-04:
That's not a different version, it's a different firmware.  A dump would be needed to support it.  You could try LightLeak or doing a serial dump.

hartzell, 2024-10-09:
I have a file that I read (`ltchiptool flash info bk7231n`; and `ltchiptool -v flash read bk7231 ghome-wp9.bin`) from a recently purchased "GHome WP9".  I can share it if it's useful, just tell me where to put it.  Never booted it into the Tuya stuff, just flashed it w/ OpenBK.

hartzell, 2024-10-09:
Here's the file [referenced above](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/699#issuecomment-2403548777): [ghome-wp9.bin.gz](https://github.com/user-attachments/files/17319535/ghome-wp9.bin.gz)

Cossid, 2024-10-09:
Thanks for supplying the dump.  Unfortunately, WP9 firmware 1.0.4 is patched, and serial flashing will be the only option for this device.

closed 2024-10-09 by Cossid

hartzell, 2024-10-10:
FWIW, flashing this via hardware is pretty straightforward.  I made [a post about it in the OpenBK forums](https://www.elektroda.com/rtvforum/viewtopic.php?p=21258299#21258299).

akj501, 2024-12-12:
> FWIW, flashing this via hardware is pretty straightforward. I made [a post about it in the OpenBK forums](https://www.elektroda.com/rtvforum/viewtopic.php?p=21258299#21258299).

Hey,

I have the UK version of this, are you able to help me with this?