tuya-cloudcutter/tuya-cloudcutter #701: DEVICE DUMP - Moes S16Pro Smart AC Controller (IR)

colino17, 2024-08-29:
Attached are a few device dumps for the Moes S16Pro Smart AC Controller (IR).

[Device Dump Download](https://github.com/user-attachments/files/16803802/s16pro.zip)

Main Module Firmware: V1.3.1
MCU Module Firmware V1.0.0

I was unable to use Cloudcutter for this, but was able to get it flashed with LTChipTool.

**A few notes:**
- In one of the pics below I've identified the pins that needed to be soldered (red=3v / green=ground / yellow=tx / pink=rx /orange=cen).
- I found that the on-screen buttons were not operational unless the device was set up via the Tuya App first. There may be a fix for this without needing the Tuya App, but what I did was set up the device, chose some random IR codes, and set my min/max temperature ranges using the App. I then deleted the device from the Tuya App and flashed it with an ESPHome firmware using LTChipTool.
- I was able to set up different IR codes in ESPHome than I had briefly set up in the Tuya App.
- [This GitHub](https://github.com/make-all/tuya-local/blob/main/custom_components/tuya_local/devices/ir_moes_heatpump.yaml) was useful when I was figuring out the datapoints.

**Sample ESPHome Yaml:**
```yaml
esphome:
  name: lounge-thermostat
  friendly_name: lounge_thermostat

bk72xx:
  board: generic-bk7231n-qfn32-tuya
  
logger:

api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 
    gateway: 
    subnet: 255.255.255.0
  ap: 
    ssid: "Lounge-Thermostat Hotspot"
    password: ""

captive_portal:

########################
### IR BLASTER SETUP ###
########################
remote_transmitter:
  id: ir_t
  pin: P7
  carrier_duty_percent: 50%
remote_receiver:
  id: ir_r
  pin: P8
  dump: all

######################
### TUYA MCU SETUP ###
######################
uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

tuya:

###############
### SENSORS ###
###############
sensor:
  - platform: tuya
    name: "Lounge Humidity"
    sensor_datapoint: 12
    unit_of_measurement: "%"
  - platform: tuya
    name: "Lounge Temperature"
    sensor_datapoint: 2
    unit_of_measurement: "°C"
    filters:
      - multiply: 0.1

#######################
### CLIMATE DEVICES ###
#######################
climate:
# THIS IS THE TUYA CLIMATE ENTITY
# IT RELAYS STATE CHANGES TO THE SECONDARY CLIMATE ENTITY WHICH PROCESSES THEM AS IR COMMANDS
  - platform: tuya
    name: "Downstairs"
    id: "climate_primary"
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step: 1
    switch_datapoint: 1
    target_temperature_datapoint: 3
    target_temperature_multiplier: 1
    current_temperature_datapoint: 2
    current_temperature_multiplier: 0.1
    supports_heat: true
    supports_cool: true
    active_state:
      datapoint: 4
      cooling_value: 0
      heating_value: 1
      fanonly_value: 3
      drying_value: 4
    fan_mode:
      datapoint: 5
      auto_value: 0
      low_value: 1
      medium_value: 2
      high_value: 3
    # AUTOMATION THAT WHENEVER THE DATAPOINTS UPDATE THOSE STATUSES SHOULD ALSO BE UPDATED ON THE MAIN CLIMATE ENTITY
    on_state:
      - lambda: |-
          if (x.mode != (id(climate_secondary).mode)) {
            auto call = id(climate_secondary).make_call();
            call.set_mode(id(climate_primary).mode);
            call.perform();
          }
          else if (x.fan_mode != (id(climate_secondary).fan_mode)) {
            auto call = id(climate_secondary).make_call();
            call.set_fan_mode(id(climate_primary).fan_mode);
            call.perform();
          }
          else if (x.target_temperature != (id(climate_secondary).target_temperature)) {
            auto call = id(climate_secondary).make_call();
            call.set_target_temperature(id(climate_primary).target_temperature);
            call.perform();
          }
      - lambda: |-
          if ((id(climate_secondary).swing_mode) != CLIMATE_SWING_BOTH) {
            auto call = id(climate_secondary).make_call();
            call.set_swing_mode(CLIMATE_SWING_BOTH);
            call.perform();
          }

# THIS THE THE SECONDARY HIDDEN CLIMATE DEVICE THAT PROCESSES THE COMMANDS FROM THE PRIMARY ENTITY AS IR COMMANDS 
  - platform: fujitsu_general
    id: "climate_secondary"
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step: 1
```

 

[image]
[image]
[image]
[image]
[image]

colino17, 2024-09-01:
For anyone stumbling across this in the future looking to use this device, I've also designed a 3D printed wall mount which may be useful:

[](https://makerworld.com/en/models/616461#profileId-540014)

Cossid, 2024-09-04:
Added as `MOES S16Pro Smart AC Controller v1.3.1`

closed 2024-09-04 by Cossid

fabianoarruda, 2024-09-21:
Thank you @colino17 for this. I've been struggling to use this device with HA, I even collaborated in the tuya_local repo that you linked above to add compatibility for this device, but it's still a pain to use. [Here is the discussion](https://github.com/make-all/tuya-local/pull/1755#issuecomment-2311695081) about the current implementation I ended up using, in case you are curious. It works but it's not very reliable since it depends on the cloud to send commands. If it's possible to use it with ESP Home instead, then that's great news! I'm interested in replacing my current setup with this. I think this device has a lot of potential because of its unique features: the touch screen with buttons, and the ability to sync commands from the original IR remote so you can use both. And the range of IR sender/receiver is really good. Too bad it's very difficult to integrate via Tuya.

colino17, 2024-09-22:
@fabianoarruda I've been using mine with ESPHome for over a month now and it's been rock solid. Theoretically it should be even easier to do since Cossid added it to Cloudcutter. When I set up mine I needed to solder pins to flash. Should be much simpler now.

Like I mentioned above the only issue I noticed was having to set it up in the Tuya app first to "activate" the touch screen buttons and the min/max settings before replacing the Tuya firmware with ESPHome.

Now it's running perfectly on ESPHome without any Tuya firmware and cloud access. I even have it on a local-only VLAN.

Feel free to ping me if you have any questions about the YAML.

colino17, 2024-09-25:
> Hi @colino17 this is amazing you mention that it should be much easier now that @Cossid added it to the Cloudcutter. Does this mean I can update the firmware "over the air". 
> 
> Do you have any instructions on how to do this I have not used Cloudcutter before.
> UPDATE: I've been reading through how the Cloudcutter exploit works and it doesn't mention anything about loading the ESPHome firmware onto the chip, is that possible? 
> It only says once the device is cut from the cloud you can use a modified TinyTuya to communicate with it whatever that means.

Yes, once exploited you can flash it with whatever compatible firmware you like. Digiblur has a YouTube video on Cloudcutter where he does exactly that. That would be a good place to start.

slnnz, 2024-09-29:
Thanks @colino17  this sounds great, just to confirm when you say to make sure you setup the device in the app first and load up a air conditioning system through the tuya app otherwise you won't have the touchscreen still working.

1. Does doing this make sure the touchscreen sends the right commands like if you setup a fujitsu in the app but then setup the unit with a Daikin using esphome the touchscreen will still work and send the right IR Codes?
2. How does it remember that you setup the remote in the first place because to get it flashed with CloudCutter you have to reset it a few times, wouldn't this lose the settings you entered in the app?

colino17, 2024-09-29:
> Thanks @colino17  this sounds great, just to confirm when you say to make sure you setup the device in the app first and load up a air conditioning system through the tuya app otherwise you won't [...]

I actually tested this exact scenario because I was confused about this aspect. I have a Fujitsu Mini Split, so I selected a random non-Fujitsu heat pump in the app. Once I flashed it to ESPHome I set it up as a Fujitsu and it worked fine.

I feel like there must be some way around needing to set it up in the Tuya app at all, but I didn't dig too deeply into it.

The way it seems to work is that the on screen buttons update "datapoints" in the Tuya MCU chip (which is a subcontroller on the board). The main chip then reads those datapoints and performs an action based on them. The main Beken chip is what holds the Tuya firmware which is flashed over with ESPHome.

Before doing the app step, the buttons didn't seem to change the datapoints. Almost like there is some low level code there that gets activated.

Once you flash the ESPHome firmware, ESPHome reads the datapoints and does whatever you tell it to do (ex: send whatever IR code you program in the ESP yaml).

slnnz, 2024-10-01:
Hi @colino17 I have managed to flash one of my devices which went pretty well with the tuya cloudcutter. Now I'm trying to install your script above into it however I'm having 2 errors with the code see below, do you have any idea why it's not working for me has the ESPhome code been updated?

[active_state_topic] is an invalid option for [climate.tuya]. Did you mean [action_state_topic], [active_state_datapoint], [mode_state_topic]?
  active_state_topic: 
    datapoint: 4
    cooling_value: 0
    heating_value: 1
    fanonly_value: 3
    drying_value: 4
  
  [fan_mode] is an invalid option for [climate.tuya]. Please check the indentation.
  fan_mode: 
    datapoint: 5

slnnz, 2024-10-01:
@colino17  thanks for the prompt reply it ended up being my ESPHome docker instance was from 2023 so well outdated. Just installed the new 2024 Sep one and it's now downloading.

fabianoarruda, 2024-10-19:
Hey @colino17 @slnnz Thanks for all the info you guys posted, very helpful. I was following the updated tutorial from Digiblur. When prompted I selected the specific profile for that device in the menu (S16Pro Smart AC Controller v1.3.1) and Checked that the firm version is the same I have (1.3.1). Then I selected `1.3.1 - BK7231N / bk7231n_common_ir_ctr_ty` and finally, `ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin`. It looks like the device is now exploited successfully, but I can't get the device to connect again to cloudcutter. Not sure if the firmware is successfully installed, It stopped at this point, Last line `wlan0: AP-ENABLED`. I waited for hours and nothing more happened. Not sure what to do next. Should I retry from the beginning? Maybe I bricked the device? :/

My guess was that I first need to load ESPHome-Kickstart, and after that, I could update it with the more customized setup based on the yaml provided by @colino17. Is my understanding correct?

[image]

Cossid, 2024-10-20:
> Hey @colino17 @slnnz Thanks for all the info you guys posted, very helpful. I was following the updated tutorial from Digiblur. When prompted I selected the specific profile for that device in the [...]

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

fabianoarruda, 2024-10-20:
Thank you very much @Cossid! I tried that earlier today and was unlucky. But it gave me a general idea of what to do. Then I found [these more specific instructions](https://www.elektroda.com/rtvforum/topic3970277.html#20658238) that finally worked for me:

> I have found that the following process works for me (assuming your device is already cut from Tuya and it has a new AP (not the original Tuya SmartLife AP)
> 
> 1. After your device gets stuck at the AP-ENABLED step - Just leave cloudcutter at the AP-ENABLED step (do not stop it, do not CTRL-C out if it - just leave it as it)
> 2. On your phone remove the location permission for the Tuya app
> 3. Start the Tuya app on your mobile phone
> 4. Manually add a new device selecting the device type you have
> 5. Put your device back into fast blinking pairing mode
> 6. In the Tuya app add your device and follow the steps (confirm that is fast blinking etc)
> 7. When it gets to the WIFI step - add cloudcutterflash as the WiFi name and abcdabcd as the password
> 8. Let the Tuya app try and add the device. When the device is added the fast blinking (pairing mode) stops
> 9. Go to your cloudcutter window and watch the logs - within a few minutes the update process will resume and the firmware will be successfully deployed to the device

So, I'm already running my custom firmware based on the yaml provided by colino17. The only difference is that I'm using `climate_ir_lg`. I still need to do some tweaking but it already works quite well, including the touchscreen buttons!

fabianoarruda, 2024-10-20:
@colino17 One thing I noticed, when the AC is turned off, and the Moes thermostat is also on off state, it keeps sending the "power off" ir command to my AC unit from time to time, like each minute or so. I keep hearing my AC beeping but it stays of. I suspect the device keeps broadcasting the DPS states, and the firmware code detects an update on the DPS 1 and sends a IR command. Did you notice that too? Any ideas on how to avoid that?

colino17, 2024-10-20:
> @colino17 One thing I noticed, when the AC is turned off, and the Moes thermostat is also on off state, it keeps sending the "power off" ir command to my AC unit from time to time, like each minute [...]

I use mine with Home Assistant so my YAML has the ESPHome API bits in there. I noticed the same behaviour until I connected it to Home Assistant. I believe that if the API can't connect to any end point it thinks something is wrong and reboots itself. If you aren't using it with Home Assistant try removing the API bits and see if that resolves the issue.

fabianoarruda, 2024-10-20:
Thanks @colino17. Yes, I'm using it with Home assistant, so this shouldn't be an issue. I'll try to investigate the second option.

fabianoarruda, 2024-10-20:
@colino17 Looks like the expression is not evaluated correctly. As you can see in the Logs, the primary ans secundary devices are set mode to OFF. but when I evaluate these values right after the lambda using `logger.log` they evaluate to `null` or empty. 

``` yaml
   # AUTOMATION THAT WHENEVER THE DATAPOINTS UPDATE THOSE STATUSES SHOULD ALSO BE UPDATED ON THE MAIN CLIMATE ENTITY
    on_state:
      
      - logger.log:
          format: " ------------------ Primary AC Mode: %s"
          args: ['id(climate_primary_criancas).mode']
      
      - logger.log:
          format: " ------------------ SECONDARY IR mode: %s"
          args: ['id(climate_secondary_criancas).mode']

      - lambda: |-
          if (x.mode != (id(climate_secondary_criancas).mode)) {
            auto call = id(climate_secondary_criancas).make_call();
            call.set_mode(id(climate_primary_criancas).mode);
            call.perform();
          }
          
   ...
```

[image]

colino17, 2024-10-20:
@fabianoarruda Maybe you could readjust the lambda automation to only run if the states don't match AND the state is not null.

Your HVAC device also might not have all the same modes and options as mine which may be throwing things off somehow.

fabianoarruda, 2024-10-23:
@colino17 I fixed it by doing a small change on your code. Instead of comparing `x.mode`  I'm using `id(climate_primary).mode`:

```
if (id(climate_primary).mode != (id(climate_secondary).mode)) {
  auto call = id(climate_secondary).make_call();
  call.set_mode(id(climate_primary).mode);
  call.perform();
}

On another topic, do you have any tips on how to open the device? I want to change the usb cable to a shorter one, but don't want to risk breaking the plastic enclosure.

colino17, 2024-10-23:
@fabianoarruda It was pretty simple. Just start at the bottom and pry up the edge and then work you way around. There are small clips all the way around that should pop open. Be careful once you get it unclipped. The wire is quite snug in the back housing so you have to slide it out carefully or you might rip it away from the board.

fabianoarruda, 2024-11-27:
@slnnz Unfortunately the cut is permanent. [More info here](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#can-i-uncut-a-device-and-connect-it-back-to-tuya)

Cossid, 2024-11-27:
> @slnnz Unfortunately the cut is permanent. [More info here](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#can-i-uncut-a-device-and-connect-it-back-to-tuya)

It's a bit more complicated than that.  That's generally true for CloudCutter, when you have no backup and can't serial flash.  When you do have a backup and can serial flash, your options are greater, but it's still complicated.

If you CloudCut, you'd need to restore both the app and the data partitions, and run the risk of conflicts, as you can run into other risks like cloning a MAC address or bumping a device out of someone elses account (but if the firmware was dumped and shared, it's probably not being used on SmartLife with regularity anyway).

But if you can restore via serial both app & data, you can reactive, and then cut again.

But really, the app is likely just setting DPIDs that could be reproduced in other ways, it just needs to be figured out.

Here are the DPID descriptors from SmartLife for this device, pay attention to the ones marked "rw", which indicates read/write, meaning you can send values to them.

```json
[{
        "code": "infared_switch",
        "dp_id": 1,
        "type": "bool",
        "value": false
    }, {
        "code": "temp_current",
        "dp_id": 2,
        "type": "value",
        "value": 0
    }, {
        "code": "target_temp",
        "dp_id": 3,
        "type": "value",
        "value": 16
    }, {
        "code": "mode",
        "dp_id": 4,
        "type": "enum",
        "value": "cold"
    }, {
        "code": "fan_level",
        "dp_id": 5,
        "type": "enum",
        "value": "auto"
    }, {
        "code": "fault",
        "dp_id": 6,
        "type": "bitmap",
        "value": 0
    }, {
        "code": "filter_reset",
        "dp_id": 7,
        "type": "bool",
        "value": false
    }, {
        "code": "filter_life",
        "dp_id": 8,
        "type": "value",
        "value": 0
    }, {
        "code": "upper_temp",
        "dp_id": 9,
        "type": "value",
        "value": 16
    }, {
        "code": "lower_temp",
        "dp_id": 10,
        "type": "value",
        "value": 16
    }, {
        "code": "temp_unit_convert",
        "dp_id": 11,
        "type": "enum",
        "value": "c"
    }, {
        "code": "humidity_current",
[... 248 lines cut ...]
        }
    }, {
        "abilityId": 202,
        "accessMode": "ro",
        "code": "ir_study_code",
        "extensions": {
            "attribute": "2048"
        },
        "name": "红外学习值上报",
        "typeSpec": {
            "type": "raw",
            "maxlen": 128
        }
    }
]
```

Might need to run some of those names through a translator.

Those are the things the app is likely setting somehow that should enable functionality in some way.
You can set up things in ESPHome as [Tuya Number](https://esphome.io/components/number/tuya) or [Tuya Select](https://esphome.io/components/select/tuya) for anything that is a number or enum, or [Tuya Switch](https://esphome.io/components/switch/tuya) for those that are bool.

Unfortunately, for the IR codes, I don't think there is anything to set string/text/bitmap via ESPHome that doesn't need custom code, but with determination, you could make the correct uart calls to do so.

Perhaps this will give someone with the device a push in the right direction for setting up this device without SmartLife.

slnnz, 2024-11-27:
Thanks @Cossid I think it may be related to one of these two options
```
        "abilityId": 16,
        "accessMode": "rw",
        "code": "status",
        "name": " 码库状态", #(Translates to Code Library Status)
        "typeSpec": {
            "type": "enum",
            "range": ["done", "run", "idle"]
        }
        "abilityId": 19,
        "accessMode": "rw",
        "code": "internet_disc_switch",
        "name": "断联开关", #(Translates to Disconnect Switch)
        "typeSpec": {
            "type": "bool"
        }
```

Can I use Tuya Select to choose between "Done" "Run" and "Idle" for the Datapoint 16.
Also if I add this to a working device will it show me what it is currently set too?
Sorry I'm still very new to ESPhome and what you can and can't do with Tuya devices.

Cossid, 2024-11-27:
To see what they are set to without setting them first, you'd either need to see if they're listed in the tuya debug boot/log output, or add a tuya sensor.  A select may be in an unknown state until it's explicitly told a state or you choose one.

As for the select once you know, yes, read the documentation linked, that is exactly what it does.  I have very limited experience with these, as I don't have very many TuyaMCU devices, because they are a pain and not as controllable as GPIO driven devices.

slnnz, 2024-11-27:
Great so I downloaded the logs of a working device and the one that failed and it looks like the DP 16 does appear to be different.  There was no sign of DP 19 so not sure about that one.
I think if I can set the DP 16 to 0 then that should maybe make it work. 

I'm guessing 0 = Done and 2 = Idle.

So using this example I will try and change it is that right?

```
# Create a select
select:
  - platform: "tuya"
    name: "Sensor selection"
    enum_datapoint: 16
    optimistic: true
    options:
      0: Done
      1: Run
      2: Idle
```

Cossid, 2024-11-27:
Correct, the enum ranges listed in the schema start at 0 and count up.

Your example looks correct.

ESPHome tuya boot/log dump only lists what has been communicated by query, but sometimes things aren't in that list until set or defined, depending on what the TuyaMCU firmware is programmed to do.  Which is why it's helpful to have the full schema list from Tuya, because that is what Tuya has designated this device should be able to communicate.

slnnz, 2024-11-27:
Well thanks everyone for their help I have successfully enabled the touchscreen and seem to have full functionality all be it with a small error in the logs every time I press a button but that seems to happen on all my working ones anyway. 

[E][tuya:297]: Invalid command (0x38) received

Also I've added the option to set the Max and Min temperatures that can be set using the touchscreen. Here's the full code block for both options. I believe it just needs to come after the tuya: line.

```
############################
### Fix the Touch Screen ###
############################

# Create a select
select:
  - platform: "tuya"
    name: "Touch Screen Setup"
    enum_datapoint: 16
    optimistic: true
    options:
      0: Done
      1: Run
      2: Idle

####################################
### Enable the Min/Max on Screen ###
####################################
number:
  - platform: tuya
    name: "Max Display Temp"
    number_datapoint: 9
    min_value: 16
    max_value: 30
    step: 1
  - platform: tuya
    name: "Min Display Temp"
    number_datapoint: 10
    min_value: 16
    max_value: 30
    step: 1
```

slnnz, 2024-11-27:
Also while I'm at it in case anyone else had a problem with certain Fujitsu heat pumps not responding to the IR Signals I found that certain Fujitsu's don't like the Swing Both option so I had to change the last lambda to the following which seems to work on all the Fujitsu's that I've tried so far. 

Swing Off seems to just set the fins/vanes to the up position for cooling and the down position for heating.

```
      - lambda: |-
          if ((id(climate_secondary).swing_mode) != CLIMATE_SWING_OFF) {
            auto call = id(climate_secondary).make_call();
            call.set_swing_mode(CLIMATE_SWING_OFF);
            call.perform();
          }
```

colino17, 2024-12-02:
> Also while I'm at it in case anyone else had a problem with certain Fujitsu heat pumps not responding to the IR Signals I found that certain Fujitsu's don't like the Swing Both option so I had to [...]

Are you sure your device has both horizontal and vertical swing? In my experience certain heat pumps only have vertical swinging fins.

slnnz, 2024-12-04:
> Are you sure your device has both horizontal and vertical swing? In my experience certain heat pumps only have vertical swinging fins.
> 
@colino17 No my units only have the vertical fins, but strangely it worked totally fine on some older units with only vertical fins but failed to work on the newer ones. I guess I could modify the code to only do Climate_Swing_Vertical but I'm happy with no swinging.

One thing that may be nice to work out would be if there is a way to manually set the swing mode in Home Assistant. 
I have a basic IR Blaster flashed with ESPHome and in its yaml file I only have the following which in home assistant gives me an option for swing with the setting Off, Both, Vertical and Horizontal. It appears at the bottom under mode and fan mode.

```
climate:
  - platform: fujitsu_general
    id: ac
    name: "Lounge AC"
    receiver_id: rcvr
    transmitter_id: tamtr
    sensor: lounge_temp
```

fabianoarruda, 2024-12-07:
> [E][tuya:297]: Invalid command (0x38) received

@slnnz I often see this error in my logs, too. So I don't think it's related to anything you have done differently. It doesn't seem to cause any problems, but it always bugged me.

> One thing that may be nice to work out would be if there is a way to manually set the swing mode in Home Assistant.
I have a basic IR Blaster flashed with ESPHome and in its yaml file I only have the following which in home assistant gives me an option for swing with the setting Off, Both, Vertical and Horizontal. It appears at the bottom under mode and fan mode.

Looking at [Tuya Climate docs](https://esphome.io/components/climate/tuya.html), there is a `swing_mode` which is optional, with `vertical_datapoint` and `horizontal_datapoint` attributes. I guess you need to set DP number to at least one of these so it will be exposed to HA. The problem is the device doesn't have such DPs.

slnnz, 2024-12-11:
> Are you sure your device has both horizontal and vertical swing? In my experience certain heat pumps only have vertical swinging fins.
No our heatpumps only support vertical swinging but it was strange that the older ones worked ok and only the newer ones had the issue.

On a different topic does anyone know how to get the Humidity to show up on the Home Assistant climate card so I can make it look like this example from a Tuya Local Climate card.
[image]

I can see how to do it using the Home assistant climate template but not sure how to do that from the esphome side. Would I need to make a custom Template climate helper?

Cossid, 2024-12-11:
Humidity is not a part of the `Climate` object in ESPHome, at least not as a read-only, only as a target.  You'd need to just add a Tuya Sensor with a DPID of 12, and and set the `device_class` of that sensor to `humidity` and hope it picks up on that, otherwise, yeah, create a helper.

fabianoarruda, 2025-07-18:
Thanks again @colino17 for the original idea. I've made some tweaks since then, and just want to share my final version, which now makes it work just like the original device with Tuya firmware (the ability to read commands from the original remote and sync back to Tuya MCU).

You can see it [here](https://github.com/esphome/esphome-devices/blob/b25213cceab215180446c96d1f80b995bc93b669/src/docs/devices/Moes-IR-Thermostat-S16Pro/index.md), I opened a [PR](https://github.com/esphome/esphome-devices/pull/1158/files) to add it to ESPHome devices list.