tuya-cloudcutter/tuya-cloudcutter #703: Couldn't flash any  of Antelia F1s202-EU my SmartPlug

gris74, 2024-09-06:
**1st try** 
```
avy@ESPHome-RPi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -s IOT Uaxtptgm69* -w wlan0
Building cloudcutter docker image
[+] Building 1.2s (13/13) FINISHED                                                                                    docker:default
 => [internal] load build definition from Dockerfile                                                                            0.0s
 => => transferring dockerfile: 462B                                                                                            0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                  0.9s
 => [internal] load .dockerignore                                                                                               0.0s
 => => transferring context: 2B                                                                                                 0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9ac27d4ecadc3ef02f980a8e2b37c7e8cdbf24039c5eddd9f98af  0.0s
 => [internal] load build context                                                                                               0.0s
 => => transferring context: 3.02kB                                                                                             0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-e  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                   0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                             0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                        0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                    0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                      0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                       0.0s
 => exporting to image                                                                                                          0.0s
 => => exporting layers                                                                                                         0.0s
 => => writing image sha256:337f5a8f8bf58be8150dc87032b5b8bc8dff62671c64046c3c5204ca20a8e4b2                                    0.0s
 => => naming to docker.io/library/cloudcutter                                                                                  0.0s
Successfully built docker image
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 ► By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Antela
   AIJIA
   Alantop
   Aldi
   Alic
   Amico
   Anko
 ► Antela
   Aoycocr
   Arlec
   Atarm
   Athom
   Atomi
   Aubess

[?] Select the article number of your device: F1s202-EU Smart Plug
   BAW534 E27 9W RGBCT Bulb
 ► F1s202-EU Smart Plug
   SPL-W-TY-PM-FR-RY Plug with Energy Monitoring CB2S v1.1.8

[?] Select the firmware version and name: 1.3.5 - BK7231T / oem_bk7231s_dltj
 ► 1.3.5 - BK7231T / oem_bk7231s_dltj

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: antela-f1s202-eu-smart-plug
Selected Profile: oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
[... 107 lines cut ...]
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/VuzHerYSG8tz.deviceconfig
Saved device config in /work/configured-devices/VuzHerYSG8tz.deviceconfig

### ================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
..
Found access point name: "SL-ANTELA-A8CA", trying to connect...
Device 'wlan0' successfully activated with '5058f21c-8d28-43fb-ac84-9de3fa3e3ca9'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

Dont know what to do, I had try 3 of my 4 plugs.
Any idea or help ?

Cossid, 2024-09-06:
1.1.17 for plugs is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.  Serial flashing will be your only option.

closed 2024-09-06 by Cossid

gris74, 2024-09-06:
@Cossid Thanks for you answer !
I didn't read it anywhere yet about plugs necessity of serial flashing  :/

3wnbr1, 2024-12-01:
If it helps, the ones on patched firmware had the following at a new inscription at the back - `HS202401`
This wasn't the case with the older ones which I flashed successfully