tuya-cloudcutter/tuya-cloudcutter #704: Connecting to wrong AP

marsp88, 2024-09-09:
Hello all,
I am trying to flash some Martin Jerry switches and they do create an AP for the Smart Life app.
However, when I run cloudcutter on a raspberry pi as described here: [LINK](https://digiblur.com/2023/08/19/updated-tuya-cloudcutter-with-esphome-bk7231-how-to-guide) tuya-convert connects to an open wifi from a eufy vacuum in my building and not my light switch. Is there a way to select what wifi to connect to?

Thank you!

Cossid, 2024-09-09:
There is no built-in way to do that with a parameter currently, but you could modify the script quickly to do so.

In [`common.sh` line 76](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/f1029fe5194cd09a1681090f39d38ada16763dc4/common.sh#L76C13-L76C38) change `SSID_REGEX="-[A-F0-9]{4}"` to `SSID_REGEX="SL-SPECIFICIDHERE"`

Cossid, 2024-09-13:
If you update your checkout, I have added a `-o` to override what AP you're trying to connect to, so `-o SL-SPECIFICIDHERE`

closed 2024-09-13 by Cossid