tuya-cloudcutter/tuya-cloudcutter #706: Carro Ceiling Fan

jeremysalwen, 2024-09-12:
I have a Carro Ceiling Fan (https://carrousa.com/collections/smart-ceiling-fan/products/akron-52-inch-3-blade-smart-ceiling-fan-with-led-light-kit-remote-silver-red-wood)  which is supposedly compatible with Tuya (https://community.home-assistant.io/t/carro-smart-home/377349).

I tried the compatibility test script, and it did seem like the AP rebooted after about 60 seconds, although there were no other external signs of "freezing" (it still responded to the remote, albeit that may be a different chip than the one that is being exploited?)

Are there any more checks I should do to make sure it will be compatible before I try to go ahead and exploit it?

Cossid, 2024-09-12:
The compatibility script is hit-and-miss with TuyaMCU devices (as you noticed, the MCU handles some functionality on it's own), and the AP has a natural timeout in most cases anyway, you should specifically be looking for the freeze of the status light, or a network drop.

But either way, that script only signifies potential, you'll need a valid profile in order to move forward.  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has finding the firmware version installed is usually the easiest step forward when there isn't a profile for you exact device.

jeremysalwen, 2024-09-12:
What does "network drop" mean in this context? There is no internet access through the AP so I'm not sure how to check that other than seeing the AP disappear and reappear (which did happen).

I checked the BSSID is "Tuya Smart Inc".  I paired it in the app and it said "Main Module: V2.1.6 MCU Module: V1.0.0"

Cossid, 2024-09-12:
You connect to the AP to send the exploit, a drop when it goes away from crashing may be visible, depending on client logging (but is subject to timeout variables).

Yes, 2.1.6 TuyaMCU is exploitable, you can choose the `by firmware version and name` option and select `2.1.6 - BK7231N / bk7231n_common_user_config_ty` and then whatever device it presents, select it, and it should work.

You'll be on your own for finding an mapping datapoints, but TuyaMCU fans are pretty easy generally.

jeremysalwen, 2024-09-12:
Okay thanks.  I'm reading through the docs, and I'm trying to understand what state it will be in after the exploit.  If I don't flash new firmware, it will basically be the original firmware, except it will have a privately generated local key and it won't phone home to the cloud -- so I could use tinytuya, localtuya, or tuya-local?

Also, is this the right place to ask questions, or is there somewhere else I should post if I encounter difficulties?

Cossid, 2024-09-12:
You don't want to do `cut` when you don't have an exact device profile, as cutting sends the DPIDs of whatever device that profile belongs to.  Cut is really only for exact device matches or known compatible ones.  The firmware exploit is not the same as device compatibility.

You'd need to specifically find a profile for a fan with 2.1.6 firmware and hope it's compatible, or create a new one (no guide on how to do that).

Here is fine for general support, though it will be limited, as we don't really support device functionality here.

Cossid, 2024-09-13:
If you want to try cutting, I guess I'd probably recommend you try as the Treatlife DS03 fan and light switch, as that should have at least the core matching DPIDs.

jeremysalwen, 2024-09-13:
I see. Since it's a TuyaMCU device, I should expect that flashing the ESP chip won't actually affect the usage of the remote control, right?  I am thinking I would probably just flash ESPHome since that seems to be the biggest community of users.  I would still need to figure out the DPID stuff, but I would feel less worried about bricking something.

Cossid, 2024-09-13:
That is the most likely assumption, but not a guarantee.

I fully expect the fan/light to share DPIDs with the Treatlife DS03, so https://devices.esphome.io/devices/TreatLife-DS03 would be a good example for the TuyaMCU/fan/light part of it.

jeremysalwen, 2024-09-15:
Okay, just want to double check I have this right.  If I just want to install ESPHome, I can use the bk7231n firmware included in the repo?   [ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/custom-firmware/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin)  And then from there I can OTA update to my final configuration?

When I'm actually doing the flashing, since it's custom firmware I just use "By firmware version and name" but only the firmware version actually matters?

Cossid, 2024-09-16:
> Okay, just want to double check I have this right. If I just want to install ESPHome, I can use the bk7231n firmware included in the repo? [...]

That is all correct.

jeremysalwen, 2024-09-16:
Okay, I gave it a go... it said "The profile you selected did not result in a successful exploit".  I'm not 100% sure what happened, because again, it's a ceiling fan without any status LED, so literally no feedback about what's happening except checking the appearance/disappearance of APs.

1. It told me to "power cycle and put my device in AP mode again", but before I could power cycle it(which involves flipping a circuit breaker in my basement), it already told me it failed.  Does the lack of power cycling mean it wasn't properly done?  Or does the fact that it was able to connect to the AP anyways mean that it's moot?
2. For a while after the exploit, I could see a GLAMP_<random_chars> access point.  I think this was just the exploit's access point, not anything generated by the device?
3. The CarroHome-<randomchars> still existed after the failed exploit and I could pair with it using the Carro app -- I'm assuming this confirms nothing was exploited?
4. The only other instructions which I'm aware I didn't follow is that I didn't have /etc/dhcpd.conf so I didn't edit it at all.

jeremysalwen, 2024-09-16:
If the fan does have an LED to indicate AP mode, it is hidden when the fan is installed on my ceiling.  I know how to put it in AP mode (hold down the button on the remote for 5 seconds), the manual also doesn't mention any LED etc for feedback on the fan: https://images.thdstatic.com/catalog/pdfImages/cc/cca37fab-c76d-44ed-b5c4-a02ae3c68ffe.pdf (it does mention an LED on the smart remote).

1. It definitely connected to the right AP, I could see in the CLI output that it connected to CarroHome-77CA both times
[image]
2. I realized GLAMP is probably the Google Lamp AP.  I think I triggered it by accidentally trying to bluetooth connect to it from the Carro app, but otherwise it didn't show up in the logs of cloudcutter.
3.  What full log would you want, beyond the screenshot I sent?

Cossid, 2024-09-16:
When I say LED light in this case, I mean the actual light on the fan, not a status LED.

The screenshot shows most of what I'd be looking for (verifying what AP was being connected to, that it reconnected to the correct one, and that the exploit ran but didn't work), but doesn't verify which profile you actually chose.

It is possible your device has a firmware version of 2.1.6 but isn't a TuyaMCU device, or has a custom-compiled device-specific firmware based on the standard TuyaMCU firmware and they just didn't give a unique version to it.

It seems like that could be the case being the LED light doesn't give any mode indication.

If that is the case, it would require a firmware dump for support to be added.  You could try LightLeak if you have an android phone and a dev board.

jeremysalwen, 2024-09-16:
I'd be happy to try dumping with LightLeak, but I'm not sure what you mean by a dev board.

Cossid, 2024-09-16:
A programmable device, something like a NodeMCU you can install custom firmware to that also has wifi.  LightLeak has pre-compiled binaries for ESP8266, ESP32, Beken (BK7231T/BK7231N), and Realteak RTL8710B.

jeremysalwen, 2024-09-16:
Would an orange pi 5 pro work?

Cossid, 2024-09-16:
It's not supported natively.  If there is a PlatformIO target for it, it might technically be possible, but you'd have to figure it out on your own (or supply a patch).

jeremysalwen, 2024-09-16:
Okay, I didn't realize how cheap a NodeMCU board is, I think I will just buy one.  If LightLeak doesn't work, I assume there is no other way to attempt an OTA exploit, and the next step would be to take the fan off my ceiling, and solder a serial connection to it, etc?

Cossid, 2024-09-16:
Correct, LightLeak is the only OTA dump method, and direct serial would be the only other option.

jeremysalwen, 2024-10-01:
I am not sure if this is the right place to discuss it, but it seems like the lightleak android app is broken and so I am unable to dump the firmware.  I am getting the symptoms exactly described here, which has been an open issue since january: https://github.com/tuya-cloudcutter/lightleak/issues/23

1. If I select Unconfigured state, it connects to the CustomAP but then says "the device" doesn't respond to pings (confusingly, since it never even told me to put the device in AP mode, nevermind tried to connect to it!)
2. If I select the "Configured to join CustomAP" state, it connects to the devices AP, but then gives the same error about the device not responding to pings.

Cossid, 2024-10-01:
I pretty much never use anything except the Unconfigured state option, the rest are a bit confusing.

It should connect to the CustomAP, then scan for and connect to a Device AP, before it even attempts to check pings.  It doesn't instruct you when to put the device into AP mode, it just expects it to be in AP mode most of the time (whenever it says `Establish a connection with the device`).  It states what AP/device name it connected to before it does any ping checks.

LightLeak isn't an easy to use or polished app, but it also isn't really under active development, so if it doesn't work for you, that is what it is.  You would need to serial dump instead.

jeremysalwen, 2024-10-01:
> It should connect to the CustomAP, then scan for and connect to a Device AP, before it even attempts to check pings.

Yes, exactly -- it's failing to do this, which indicates something is severely wrong with the app.  Forget about the other options, just following exactly the [instructions in the guide](https://github.com/tuya-cloudcutter/lightleak) and selecting Unconfigured:  It connects to LightLeakIdle and then fails the ping check immediately:

[image]

This indicates a bug, right?

Cossid, 2024-10-01:
Possibly a lack of permissions or incompatible with an Android version.  That shows it's not even configuring LightLeakCustom, just that it connected to LightLeakIdle, then fails

Reminder that permissions must be explicitly set manually for side-loaded apps.  I've only used LightLeak on Android 11, not sure if there are requirements around android version.

I guess I wasn't correct on step names

`Establish connection with the device` is for connecting to LightLeakIdle and setting up LightLeakCustom

`Connect to smart device WiFi` is where it expects the target device to be in AP mode.

jeremysalwen, 2024-10-01:
Ok thanks! I think it was due to incompatibility with Android 9.   Using a phone with Android 14 worked (and it asked for permissions on its own!)  Well, by worked, I mean the app was functional, sadly the Carro ceiling fan did not seem to be vulnerable to any of the Lightleak BK7231N profiles.

I will try a serial dump next (once I figure out how to get it powered when off the ceiling). Thanks for the help so far.

jeremysalwen, 2024-10-01:
Ah so it looks like the chip is a WBR3?  Which means [it is not supported?](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/655)  I'm surprised it was dropping off the network when I was testing it.

[image]

Cossid, 2024-10-02:
Realtek has the same software flaw/vulnerability, which is why it freezes the same way, however, the exploit pattern has not been researched, and there is no 3rd party firmware available for RTL87x0C yet (LibreTiny has partial, but incomplete support).

Closing this, as there is currently nothing further to pursue in regards to Tuya-CloudCutter.

closed 2024-10-02 by Cossid

Cossid, 2025-11-25:
This device is now supported as
`Carro CES523A2L1 Ceiling Fan v2.1.6` and `Carro CES523A2L1 Ceiling Fan v2.5.2`