tuya-cloudcutter/tuya-cloudcutter #707: Energizer RGB+2700-6500K Bulbs - A60 E27 + C37 E14 - UK

karldonteljames, 2024-09-13:
Hi All,
Got this a couple of weeks ago, tried tuya-convert and kinda gave up until I found this yesterday.

I've tried a few different firmware versions, but I'm missing something or these devices have been patched. I added them to both the Tuya and Energizer app (to check they reported the same) both apps are reporting the firmware version 1.5.21

I have 2 bulbs and they have the following markings on them:
GLS ES Bulb-
Model-S1600(364116)
8.5W 220-240V-50/60Hz A60 E27
806LM RGB+2700-6500K 219962
MADE IN CHINA
 
Candle SES Bulb - 
Model-S16901(364117)
4.8W 220-240V-50/60Hz C37 E14
450LM RGB+2700-6500K 219962
MADE IN CHINA

I've stripped down the GLS and the images are attached. I have an FTDI adapter, but I have no idea how to get a firmware dump or how to connect the pins.

Any help would be appreciated.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Cossid, 2024-09-13:
1.5.21 is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.  Serial flashing will be the only option for this device.

CB2L is not an easy flash and requires pads on the back side of the card to accessed: https://docs.libretiny.eu/boards/cb2l/#pinout (RX1/TX1), and they are sometimes not exposed due to mounting.

closed 2024-09-13 by Cossid

karldonteljames, 2024-09-13:
Thanks for this. Just ripped off the control board, and it's a BK7321N.
I don't want to take each of the bulbs apart and solder them, looking for a handful of SES, ES and bayonet bulbs, and about 20ish GU10's but not sure on the best route now.

Thanks for getting back to me so quickly.