tuya-cloudcutter/tuya-cloudcutter #717: Bin file.

chatz86, 2024-10-01:
Hi guys,

Please find attached a copy of the dump file 
[dump_1921681115.bin.zip](https://github.com/user-attachments/files/17216211/dump_1921681115.bin.zip)
ump from the cloud cutter android application.

How do j go about creating a firmware?

Cossid, 2024-10-02:
This is a 1KB file, which is not a valid dump.  Additionally, details of what the device is would be needed.

closed 2024-10-02 by Cossid

chatz86, 2024-10-02:
Hi there,

Its a Mirabella Genio UFO ceiling light which is compatible with Tuya. MCU
v3.3.35

https://www.mirabellagenio.com.au/product-range/mirabella-genio-wi-fi-dimmable-19w-bc-rgbcct-ufo-led-bulb/

Maybe I didn't do something right. I'll have another play. I did open one
of the units. It contains an esp8285 with a TYLC5.

I've attached a couple of images as well. Any help would be greatly
appreciated.

Best Regards,

Chat Jayawardana

Cossid, 2024-10-02:
ESP8285 is an ESP8266 with built-in flash.  We won't be able to give you information on compiling a working firmware, but Tasmota or ESPHome would be capable of running on the device, but finding the valid working configuration would be up to you to find.

CloudCutter does not work on ESP8285/ESP8266 devices.