tuya-cloudcutter/tuya-cloudcutter #720: Suggestions for this script to work with Manjaro

TechKnowCase, 2024-10-12:
Hello,

I'm not a  developer but I have a few suggestions for this to work with Manjaro, meaning the modifications I had to do be successfully tuya-cloudcutting. I understand that you made this with with the objective to be running it in debian (or Raspbian) but this could be compatible with those 2 fix:

the first thing I had to do is to modify the script to change the "service" command to the systemctl appropriate equivalent. My suggestion: is there an if statement that can be used to check if "service" or "systemctl" is used?

second, I had to turn off Apparmour (#692)  for the script to continue. Again, if there's a way to check if it's running and then running the "sudo aa-teardown" command? [Souce](https://wiki.archlinux.org/title/AppArmor#Disabling_loading)

Cossid, 2024-10-12:
Debian & Raspbian are both systemd, so systemctl exists there, so we can simply just switch to that.

As for apparmour, can you try adding the following to a script and seeing if it works correctly?

```bash
  if type "aa-enabled" &> /dev/null && aa-enabled | grep -qw Yes; then
      echo "Detected app armour"
      echo "This has been known to block hostapd, which is required to complete the exploit"
      read -p "Do you wish to stop the app armour service? [y/N] " -n 1 -r
      echo
      if [[ $REPLY =~ ^[Yy]$ ]]; then
          sudo aa-teardown
      fi
  fi
```

If it does, we can add that to safety_checks.

kuba2k2, 2024-10-13:
Just note that systems that aren't based on systemd (such as OpenRC/Alpine Linux) don't have the `systemctl` command, but do have `service`.

Cossid, 2024-10-13:
True, but systemd has become pretty standard across most distros like Debian, RedHat, and Arch.  There are probably more distros that support `systemctl` than `service`.

If someone wants to write a patch that safely checks for either, it would certainly be accepted though, but for now, it is switched to `systemctl` by default.

TechKnowCase, 2024-10-20:
@Cossid 
> As for apparmour, can you try adding the following to a script and seeing if it works correctly?

Thank you for your patience, Yes, It worked successfully. May I suggest to echo a message to the user that app armor isn't re-enabled till the user restart their computer?

```
[techknowcase@iUseArchBTW ~]$ sudo aa-status
apparmor module is loaded.
80 profiles are loaded.
80 profiles are in enforce mode.
   /usr/bin/akonadiserver
   /usr/bin/lxc-copy
   /usr/bin/lxc-start
[...]
   zgrep
   zgrep//helper
   zgrep//sed
0 profiles are in complain mode.
0 profiles are in kill mode.
0 profiles are in unconfined mode.
6 processes have profiles defined.
6 processes are in enforce mode.
   /usr/bin/akonadiserver (2495) 
   /usr/bin/avahi-daemon (944) avahi-daemon
   /usr/bin/avahi-daemon (992) avahi-daemon
   /usr/bin/dnsmasq (1447) dnsmasq
   /usr/bin/dnsmasq (1448) dnsmasq
   /usr/bin/mariadbd (2505) mariadbd_akonadi
0 processes are in complain mode.
0 processes are unconfined but have a profile defined.
0 processes are in mixed mode.
0 processes are in kill mode.

[techknowcase@iUseArchBTW ~]$ ./tuya-cloudcutter.sh 
Detected app armour
This has been known to block hostapd, which is required to complete the exploit
Do you wish to stop the app armour service? [y/N] y
Unloading AppArmor profiles

[techknowcase@iUseArchBTW ~]$ sudo aa-status
apparmor module is loaded.

```
The last line states that apparmor is loaded, but no profiles are active (compared to earlier).

Cossid, 2024-10-21:
Thanks for verifying.  I have added it to safety_checks, and added a line about it being disabled until turned back on or OS reboot.

closed 2024-10-21 by Cossid