tuya-cloudcutter/tuya-cloudcutter #724: Wipro Garnet 9W E27 Smart Bulb NS9500

redleo1981, 2024-10-17:
Wipro Garnet 9W E27 Smart Bulb NS9500 (Bought from Amazon.in)

Module - CB2L
MCU	BK7231N
[ltchiptool_bk72xx_2024-10-17_05-44-42.zip](https://github.com/user-attachments/files/17405163/ltchiptool_bk72xx_2024-10-17_05-44-42.zip)

Cossid, 2024-10-17:
This is firmware 1.5.21 which is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.  Serial flashing will remain the only option for this device.

closed 2024-10-17 by Cossid

redleo1981, 2024-10-18:
> This is firmware 1.5.21 which is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list. Serial flashing will remain the only option for this device.

Appreciate your help. Thanks for your update...