tuya-cloudcutter/tuya-cloudcutter #727: Request for Miboxer WL5 5 in 1 LED Strip Controller

tlacolley, 2024-10-28:
Hello and thank you for your job. 
I have several controller Miboxer L5 5 in 1 LED Strip Controller: 
https://miboxer.com/product/5-in-1-led-strip-controller-wifi2-4g

The device use the app Tuya smart. In the app I see two version Main module V2.1.6 and MCU Module: V1.0.1 
The chips is BK7231N. 

Maybe the software already integrates the device, but I'm not sure, and I don't want to make a mistake 😄 
Is one of this profile ? : 

1.0.1 - BK7231N / bk_light_rgbcw_vpylfsrdspscrycm
1.0.1 - BK7231N / cuco_switch_sw5_bk_n_com
1.0.1 - BK7231N / oem_bk7231n_strip_ffc_mk_ty

2.1.6 - BK7231N / bk7231n_common_user_config_ty
2.1.6 - BK7231N / oem_bk7231n_async_colorful_string_ty

Thank you for your help 😃

Cossid, 2024-10-28:
This device is already supported, you can do by Mfg / Model instead of version, MiBoxer, FUT039W RGBCT LED Strip Controller, which is the 2.1.6 - BK7231N / bk7231n_common_user_config_ty profile.

closed 2024-10-28 by Cossid

tlacolley, 2024-11-01:
Ok, maybe I'm dummy, but I never find the device when is scanning for Tuya  app. 

I try : 
- Mfg / Model , MiBoxer, FUT039W RGBCT LED Strip Controller /  2.0.2 - BK7231N / bk7231n_common_user_config_ty . there is no other option. 
- By firmware with the both 2.1.6 
       2.1.6 - BK7231N / bk7231n_common_user_config_ty
       2.1.6 - BK7231N / oem_bk7231n_async_colorful_string_ty
- By custom profile with  miboxer-fut039w-rgbct-led-strip-controller

Every time it's searching but nothing appear. 
The controller is in AP mode. Because when I try to with my phone, it's find it. 
The both device (Raspberry Pi 5 8go and the controller) are nearby. I am following this YT [tutorial ](https://www.youtube.com/watch?v=VFsuza3UAhk&t=1507s) and the GitHub instruction. 
My dhcpcd.conf   and NetworkManager.conf are right. 

I will try to reinstall Debian on Pi5 device, but I think my mistake is not on the installation. 
I have no more clue, if you have one it will be with pleasure

Thanks for your time.

Cossid, 2024-11-01:
Your phone is using Bluetooth more than likely, not AP mode.  Whatever you're doing to put it in pairing mode, within 10 seconds of having done so, do the same thing again, and it should put out an AP.

tlacolley, 2024-11-01:
Yes, you're right, for Bluetooth. I will try again, still I find it. But it's been two night I try. I will push longer the button.
Thanks