tuya-cloudcutter/tuya-cloudcutter #729: build_profile.py - Unknown pattern - Lumary-WifiController_201202-RGB-WW-CW

chris173972, 2024-11-05:
Upon triggering py .\build_profile.py .\Lumary-WifiController_201202-RGB-WW-CW.bin I got this error:

[+] Processing file='.\\Lumary-WifiController_201202-RGB-WW-CW.bin' as Lumary-WifiController_201202-RGB-WW-CW
RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xdd20]
                extracted to C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\Lumary-WifiController_201202-RGB-WW-CW
Missing app RBL container. Using a scan pattern instead
        0x11000: app - [NO RBL, size=0xc69e0]
                extracted to C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\Lumary-WifiController_201202-RGB-WW-CW
Storage partition:
        0x1ee000: 32 KiB - 11 keys
        - 'gw_bi'
        - 'gw_di'
        - 'gw_wsm'
        - 'is_stride'
        - 'tls_ca_cnt'
        - 'user_param_key'
        - 'mf_test_close'
        - 'gw_ai'
        - '000003r1tl'
        - 'timer_arr'
        - 'em_sys_env'
                extracted all keys to C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\Lumary-WifiController_201202-RGB-WW-CW\Lumary-WifiController_201202-RGB-WW-CW_storage.json
Storage area `user_param_key`:
        - found! Extracted to C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\Lumary-WifiController_201202-RGB-WW-CW\Lumary-WifiController_201202-RGB-WW-CW_user_param_key.json
[+] Searching for known exploit patterns
Traceback (most recent call last):
  File "C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\build_profile.py", line 45, in <module>
    haxomatic.run(app_file)
  File "C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\haxomatic.py", line 234, in run
    walk_app_code()
  File "C:\Users\Chris\OpenSource\tuya-cloudcutter\profile-building\haxomatic.py", line 141, in walk_app_code
    raise RuntimeError('Unknown pattern, please open a new issue and include the bin.')
RuntimeError: Unknown pattern, please open a new issue and include the bin.

[Lumary-WifiController_201202-RGB-WW-CW.zip](https://github.com/user-attachments/files/17635901/Lumary-WifiController_201202-RGB-WW-CW.zip)

chris173972, 2024-11-05:
[image]

Cossid, 2024-11-05:
This isn't a dump of the Tuya firmware, but rather a dump after OpenBeken has already been flashed.  A profile cannot be generated from that.  We would need a dump of the Tuya firmware.

closed 2024-11-05 by Cossid

Cossid, 2024-11-05:
Because storage was present, and this matched an existing profile, I have built a new one manually, but the schema is unverified, as it belongs to another device, so for anyone using this with the cut profile, the white channels may are may not work.

Added as `Lumary 201202 RGB+WW+CW LED Controller v1.1.6`