tuya-cloudcutter/tuya-cloudcutter #731: Add support for Star Projector SK20 on Tuya CBU module.

danishru, 2024-11-08:
I bought a new Star Projector SK20. 
Here is its dump.
[ltchiptool_bk72xx_2024-11-05_14-44-59.zip](https://github.com/user-attachments/files/17680021/ltchiptool_bk72xx_2024-11-05_14-44-59.zip)
[image]
[image]
[image]
[image]
[image]

And I wanted to flash it as in this article https://github.com/M4GNV5/esphome-SK20-Nebula-Light. But I couldn't even switch the device to AP mode. I tried many times and in different ways to turn it on, but I didn't succeed. After searching for additional information, I found that this behavior is not only mine, but also other people, for example here https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/668. As a result, I decided to reflash it via a cable. Having opened the projector, I found that it is on the [Tuya CBU module](https://docs.libretiny.eu/boards/cbu/), and has a different schematic board. I couldn't read the firmware without unsoldering the module, perhaps I made a mistake and did not swap the TX - RX pins as in the ltchiptool documentation. As a result, I unsoldered the module and was able to read the installed firmware, and successfully reflashed it to the basic ESPHome firmware. Now I am creating a firmware config for the excellent operation of this device.

The installed firmware on the device in SmartLife was "Main module: V2.0.2", "Microcontroller module: V2.0.2"
[image]

I think without disassembling it is possible to determine that you have a model on the Tuya CBU chip by the button LED, it always lights up when the power is on, and the device cannot switch to AP mode, it only blinks with a fast signal. If you disassemble the device, it looks like this.
[image]
[image]
[image]
[image]

It would be great if, based on the firmware analysis, you could tell how to switch the device to AP mode, and whether it is possible to flash it via Tuya Cloudcutter. Well, if not, then let the community know about it.

Cossid, 2024-11-08:
Added as `Tuya Generic SK20 Star Projector v2.0.2`

This device appears to requires either 10x power cycles or a 10 second button press to change between pairing modes, unsure which.

closed 2024-11-08 by Cossid

Flumeded, 2024-11-10:
So I have SK23, and it also running CBU.
I had to solder my UART/TTL adapter and change some pins.
My version of software mostly works, but I have a feeling that not everything is ported properly.

https://github.com/Flumeded/esphome-SK23-Nebula-Light

Visually your board looks the same as mine in SK23, so firmware might be the same, as well as pinout

danishru, 2024-11-10:
> So I have SK23, and it also running CBU. I had to solder my UART/TTL adapter and change some pins. My version of software mostly works, but I have a feeling that not everything is ported properly. [...]

I looked at your yaml file. Yes, I have the same pinout as you. And what about your reset button, what exactly does not work? I am currently working on an extended firmware for my SK20. I have implemented such functionality as additional scenes, support for button multiclick, with sending events to Home Assistant (such as single, double click, hold and release hold), support for a locally running timer with the ability to configure from Home Assistant, support for restoring the last selected scene and additional sensors. When I finish, I will create a repository and post my developments.

ajfriesen, 2024-12-16:
AP mode worked for me with:

1. Power
2. 7 second hold until it is off
3. 7 second hold again

source:
https://community.home-assistant.io/t/tuya-smart-galaxy-projector/354691/19?u=ajfriesen