tuya-cloudcutter/tuya-cloudcutter #732: wlp3s0 wlan failure?

BearSF, 2024-11-10:
I'm using my retro ThinkPad W520 to do this, running Ubuntu. It's fully updated, and the physical WiFi switch is turned on, and networks are visible, though as the directions explain, I am connected via ethernet.

After running the script, I get this at the end:

`================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
================================================================================

Using WLAN adapter: wlp3s0
Configuration file: /dev/stdin
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlp3s0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlp3s0: interface state UNINITIALIZED->DISABLED
wlp3s0: AP-DISABLED 
wlp3s0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlp3s0 wasn't started

`

The script waits there until I terminate it.

So is my WiFi just incompatible, or do I need to correct it with some commands? If so, what commands?

Cossid, 2024-11-10:
Are you running with `sudo`?  Was there a warning earlier in the process?

Run `nmcli -f wifi-properties device show wlp3s0 | grep WIFI-PROPERTIES.AP` and see if your device supports AP mode

Cossid, 2025-01-09:
Closing as stale.

closed 2025-01-09 by Cossid