tuya-cloudcutter/tuya-cloudcutter #733: Moes WM-104-M and WM-104B-M Device Dumps

AdamKGoddard, 2024-11-10:
[ltchiptool_bk72xx_2024-11-10_18-06-03_Moes_WM-104B-M.zip](https://github.com/user-attachments/files/17690668/ltchiptool_bk72xx_2024-11-10_18-06-03_Moes_WM-104B-M.zip)
[ltchiptool_bk72xx_2024-11-10_18-42-18_Moes_WM-104-M.zip](https://github.com/user-attachments/files/17690669/ltchiptool_bk72xx_2024-11-10_18-42-18_Moes_WM-104-M.zip)

Please see attached. Single and Dual Relay modules using Beken/Tuya CB2S modules. Tried a few cloudcutter firmwares but didn't seem to be able to use an exploit. Reverted to LTChipTool to dump and then program with ESPHome. There's a physical button and buzzer but it wasn't picked up and I haven't bothered to check which pins they're connected to because they will be behind switchplates. I also have the Single and Dual Dimmer modules and will attempt to dump/convert these as well as I had similar experience in not getting the exploit to work with them as well.

AliExpress listing example: https://www.aliexpress.com/item/1005007174345765.html

ESPHome relay configs below:

Single:
"""""""
binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P26
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
  - platform: gpio
    id: binary_switch_all
    pin:
      number: P11
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin: P8
"""""""

Dual:
"""""""
binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P26
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
  - platform: gpio
    id: binary_switch_2
    pin:
      number: P24
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_2
  - platform: gpio
    id: binary_switch_all
    pin:
      number: P11
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
        - switch.toggle: switch_2

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin: P8
  - platform: gpio
    id: switch_2
    name: Relay 2
    pin: P6
"""""""

AdamKGoddard, 2024-11-10:
Forgot to mention - Firmware in Tuya app for both devices:
Main Module: V1.3.10, Other type: V102

Cossid, 2024-11-10:
Switch firmware v1.3.10 is already on the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware).  Serial flashing will remain the only option for these devices.

closed 2024-11-10 by Cossid