tuya-cloudcutter/tuya-cloudcutter #736: Device Dump for Unsupported FEIT Smart Dimmers Costco ITM 1734165

BradFitz99, 2024-11-19:
I'm new to this
There wasn't a supported device profile for my FEIT Smart Dimmers so I used the wire method with Itchiptool.
I gathered as much information as I could but I'm  not sure where to go from here to generate a UF2 file to load into the device.
I've attached the following
- Read Flash File
- Storage DATA JSON
- Show full response
- Additional DATA passed to UPK2ESPHome
- Bottom Window Schema Download Tab
- Schema Tab Screenshot

[ltchiptool_bk72xx_2024-11-18_20-47-21.bin.dmp](https://github.com/user-attachments/files/17808517/ltchiptool_bk72xx_2024-11-18_20-47-21.bin.dmp)
[Storage DATA JSON.txt](https://github.com/user-attachments/files/17808522/Storage.DATA.JSON.txt)
[Show full response.txt](https://github.com/user-attachments/files/17808524/Show.full.response.txt)
[Additional DATA passed to UPK2ESPHome.txt](https://github.com/user-attachments/files/17808527/Additional.DATA.passed.to.UPK2ESPHome.txt)
[Bottom Window Schema Download Tab.txt](https://github.com/user-attachments/files/17808529/Bottom.Window.Schema.Download.Tab.txt)
[image]

Cossid, 2024-11-19:
This is firmware version 2.1.17 which is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware).  Serial flashing will be the only option for devices with this firmware version.  If SmartLife is requesting to upgrade, you should deny the request.

closed 2024-11-19 by Cossid

Cossid, 2024-11-19:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#what-custom-firmware-options-are-available