tuya-cloudcutter/tuya-cloudcutter #737: New(-ish) version of Tuya Generic AP-SMT-BREAKER02-1CH changed onboard chip. Device profile no longer works.

raptorecki, 2024-11-24:
Hey,

Just a heads up. It seems that _tuya-generic-ap-smt-breaker02-1ch-smart-relay_ profile is no longer effective against new(-ish) versions of this generic device. It used to have BK7231N but now it's something like this: T34 2422GWMYK X3I(1?)CPC

[image]

What is it anyway? Couldn't find any info about this chip online...

```
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 ► By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Tuya Generic
   TCP
   TEQOOZ
   TOMZN
   tongou
   Topgreener
   Treatlife
 ► Tuya Generic
   Ultrabrite
   UltraPro
   UME
   US Solid
   Ustellar
   Wipro

[?] Select the article number of your device: AP-SMT-Breaker02-1CH Relay
   809WT Motion Sensor v1.0.0
   80A Energy Meter
   812WT Motion Sensor with Alarm v1.0.2
   898WT Water Leak Sensor
   A228903 30W RGBCT Ceiling Light v1.0.2
   A60 RGBCT Bulb 1.1.2
 ► AP-SMT-Breaker02-1CH Relay
   AP-SMT-Breaker02-1CH Smart Relay
   ATS01L-US 1 Gang Touch Switch
   ATS02L-US 2 Gang Touch Switch
   ATS03L-US 3 Gang Touch Switch
   AW201 Contact Sensor
   BHT-002GB WIFI Thermostat

[?] Select the firmware version and name: 1.1.6 - BK7231T / oem_bk7231s_control_switch
 ► 1.1.6 - BK7231T / oem_bk7231s_control_switch

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Occupied by apache2 with PID 834.
Port 80 is needed to answer HTTP requests
Do you wish to terminate apache2? [y/N] y
Attempting to stop apache2.service

Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: OpenBeken-v1.17.697_bk7231t.ug.bin
   ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 ► OpenBeken-v1.17.697_bk7231t.ug.bin

Selected Device Slug: tuya-generic-ap-smt-breaker02-1ch-relay
Selected Profile: oem-bk7231s-control-switch-1.1.6-sdk-1.0.6-40.00
Selected Firmware: OpenBeken-v1.17.697_bk7231t.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
....
Found access point name: "SmartLife-AADA", trying to connect...
Device 'wlan0' successfully activated with 'feb588da-d35e-4038-868b-bfe963820af5'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/VKHS1RIylO6x.deviceconfig
Saved device config in /work/configured-devices/VKHS1RIylO6x.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-AADA", trying to connect...
Device 'wlan0' successfully activated with 'feb588da-d35e-4038-868b-bfe963820af5'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

```
{"uuid": "VKHS1RIylO6x", "auth_key": "xIkCDJz3EllzusVT", "local_key": "IZ0uZtXWfGDHYbaI", "sec_key": "uGYE8Jro8X7egVnq", "device_id": "EHp3QfcxFzErceGUySiS", "chip_family": "BK7231T", "profile_name": "1.1.6 - BK7231T / oem_bk7231s_control_switch", "device_slug": "tuya-generic-ap-smt-breaker02-1ch-relay"}
```

Cheers

Cossid, 2024-11-24:
T34 is just an internal partner/code for BK7231N I believe.  That means this one just has a different version of firmware, and the likelyhood is pretty high that it is also patched.  Nothing we can do without a firmware version or firmware dump.

closed 2024-11-24 by Cossid

raptorecki, 2024-11-27:
Thanks. I flashed it over UART. If anyone ever needs it, here's the original dump.

[readResult_BK7231N_QIO_2024-26-11-23-49-23.zip](https://github.com/user-attachments/files/17940966/readResult_BK7231N_QIO_2024-26-11-23-49-23.zip)

Cheers!