tuya-cloudcutter/tuya-cloudcutter #739: Cannot flash W509Z1 light bulb

felipecrs, 2024-11-30:
Hi there. Maybe you can help me.

This is my device:

LED light bulb W509Z1 20W 110V

https://aliexpress.com/item/1005007177628485.html

[image]

The "Main module" firmware in the Tuya app says 1.0.0 (as well as "MCU module").

I have tried the following profiles:

- bk-light-lm96xe1-16w-1.0.0-sdk-1.0.2-40.00
- bk-light-lm99xe1-16w-1.0.0-sdk-1.0.2-40.00
- oem-bk7231n-light5-db-a19-1.0.0-sdk-2.3.1-40.00
- oem-bk7231n-light5-db-a21-1.0.0-sdk-2.3.1-40.00
- oem-bk7231n-light-ty-1.3.21-sdk-2.3.3-40.00

They all failed with:

```
Found access point name: "SmartLife-E68C", trying to connect...
Device 'wlan0' successfully activated with '7cd7ad6e-f3b3-4300-8092-2038f44ee52c'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

Do you have any other idea?

Cossid, 2024-11-30:
No point in trying any profiles where the firmware version doesn't match.

Sounds like this device likely has a custom firmware, and would require a firmware dump to be supported.

closed 2024-11-30 by Cossid

felipecrs, 2024-11-30:
Got it. To get a firmware dump, would I need to get access to the programming pins?

Another question: should I keep trying for all the other 1.0.0 profiles?

felipecrs, 2024-11-30:
Also, do you think it's worth trying LightLeak?

felipecrs, 2024-11-30:
Ok, I found out that my bulb uses a LN882HKI chip. Maybe that's why it's not working through Cloudcutter (I don't see this chip being listed in any profile).

[image]

Cossid, 2024-11-30:
LN882HK is not supported by CloudCutter or LightLeak.

felipecrs, 2024-11-30:
Alright. Thank you very much.

divadiow, 2024-12-01:
Afaik OpenBeken is currently your only option on LN882H chips

felipecrs, 2024-12-01:
Right. I just don't know if it's worth the trouble for such cheap light lol. Being that cheap, it's probably going to stop working soon anyway.

FYI I found a guy who did the whole thing with this light. He even uploaded a firmware dump. OpenBeken is indeed compatible.

https://www.elektroda.com/rtvforum/topic4037269-30.html#21264807

May be useful if someone else finds this issue.