tuya-cloudcutter/tuya-cloudcutter #742: TOMZN TOB9-VAP 1-63A Adjustable Circuit Breaker

felipecrs, 2024-12-05:
Hi, I have one of these.

It comes with firmware 1.0.15, but none of the profile worked.

I took it apart and it has a CBU (BK7231N).

I will manage to get a firmware dump, but it's not being easy without soldering.

I just wonder: wasn't another profile for the same chip supposed to work?

This one: https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/profiles/oem-bk7231n-slide-strip-ty-1.0.15-sdk-2.3.1-40.00.json

Cossid, 2024-12-05:
1.0.15 is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.

closed 2024-12-05 by Cossid

felipecrs, 2024-12-05:
Thank you. Should not I even bother dumping the firmware then?

Cossid, 2024-12-05:
It never hurts to have a backup, it's just not useful to CloudCutter in this case.  It could also potentially help you determine a configuration with analyzing tools like UPK.

felipecrs, 2024-12-06:
FWIIW: https://github.com/esphome/esphome-devices/pull/942

felipecrs, 2024-12-06:
> It never hurts to have a backup, it's just not useful to CloudCutter in this case. It could also potentially help you determine a configuration with analyzing tools like UPK.

I captured the dump by the way. But UPK2ESPHome was only able to figure out the RX1/TX1 pins. The rest of the pins I found from [some other guy online](https://www.elektroda.com/rtvforum/topic3934580.html).

Let me know in case you'd like the dump to check why UPK2ESPHome didn't figure out the rest.

Cossid, 2024-12-06:
I'm pretty sure we already have this dump.  UPK2ESPHome isn't meant to be a full solution, just a starting point.  It is using configuration data stored by Tuya, but not all devices have this, of if it's not there the tool just can't do anything about it.  That's why things like device pages also exist, for more in-depth and refined solutions.

felipecrs, 2024-12-06:
Well, I do see mentions of the other pins in the extracted data:

```jsonc
    "user_param_key": {
        "bt1_pin": 17, // button (NOT DETECTED)
        "netled1_pin": 15, // status led (DETECTED)
        "rl_off1_pin": 26, // bridge reverse (NOT DETECTED)
        "rl_on1_pin": 24, // bridge forward (NOT DETECTED)
    },
```

[TOMZ-TOB9-VAP.source.json](https://github.com/user-attachments/files/18041569/TOMZ-TOB9-VAP.source.json)

felipecrs, 2024-12-06:
And this is what it generated:

```yaml
status_led:
  pin:
    number: P15
    inverted: true

uart:
  id: uart_bus
  tx_pin: TX1
  rx_pin: RX1
  baud_rate: 4800
  stop_bits: 1

sensor:
  - platform: bl0942
    uart_id: uart_bus
    current:
      name: BL0942 Current
    voltage:
      name: BL0942 Voltage
    power:
      name: BL0942 Power
      filters:
        multiply: -1
    energy:
      name: BL0942 Energy
    frequency:
      name: BL0942 Frequency
      accuracy_decimals: 2
```

Cossid, 2024-12-06:
You're welcome to submit a pull request or issue on UPK2ESPHome, but there is no guarantee the work will be completed by others.  Even when UPK isn't able to generate the complete config, if unused data was present, it can still be manually reviewed.  Tuya data is sometime too inconsistent to be able to handle every edge case.  Again, it's not meant, and never will be a complete generation tool, it's meant to be a starting point.

felipecrs, 2024-12-06:
Hey, it's alright. I totally understand that. I'm not demanding anything. I'm just trying to help with the information I have. :)

I'll open an issue there for completeness. Maybe no one will implement, but it's better to have it anyway.

Thanks a lot for all the help.