tuya-cloudcutter/tuya-cloudcutter #744: New LSC (Battletron) device

oliv3r, 2024-12-06:
I just purchased a battletron led strip from action; https://www.action.com/nl-nl/p/3206528/battletron-bureau-led-lichtstrip-razer-chroma/

I've opened up the controller, and found the following PCB

[image]
[image]

Connecting to UART2_TX I get the following:
```
V:BK7231N_1.0.1
REG:cpsr     spsr     r13      r14
SVC:000000D3          00401C1C 000033AC
IRQ:000000d2 00000010 00401e0c 301005b2 
FIR:000000d1 00000010 00401ffc 92820353 
SYS:000000df          0040192c 00000158 
ST:00000000
J 0x10000
bk_misc_init_start_type 0 0
[MEM DBG] heap init-------size:163128 addr:0x4082c8---------
[Flash]id:0xeb6015
[load]bandgap_calm=0x75->0x35,vddig=4->5
[bk]tx_txdesc_flush
tkl_ethernetif_init
tkl_ethernetif_init
tkl_ethernetif_output: invalid vif: 255!
tkl_ethernetif_output: invalid vif: 255!
[01-01 00:00:00 TUYA N][lr:0x4eb2b] reset info -> reason is 0
[01-01 00:00:00 TUYA E][lr:0xc9c73] uf_open 10 err 8
[01-01 00:00:00 TUYA E][lr:0x54b23] uf file 10 can't open and read data!
[01-01 00:00:00 TUYA E][lr:0x4f5db] ledsum data read error!
[01-01 00:00:00 TUYA N][lr:0x51923] ReadLedSumData error,use deafult ledSum:36
[FUNC]intc_init
[FUNC]calibration_main
gpio_level=0,txpwr_state=0
get rfcali_mode:0
DPLL Unlock
[01-01 00:00:00 TUYA E][lr:0x4e01d] ws2812b_led_count_set error!
[01-01 00:00:00 TUYA N][lr:0x49335] flashAppData.bSwitch:1
[01-01 00:00:00 TUYA N][lr:0x58efb] local music need not init!
[01-01 00:00:00 TUYA E][lr:0xc9c73] uf_open 10 err 8
[01-01 00:00:00 TUYA E][lr:0x54b23] uf file 10 can't open and read data!
[01-01 00:00:00 TUYA E][lr:0x4f5db] ledsum data read error!
mode:0, rate:6600000
spi_master:[CTRL]:0x00c30930 
[CTRL]:0x00c20900 
[CONFIG]:0x00000000 
[01-01 00:00:00 TUYA E][lr:0x50cd1] lig way err
[01-01 00:00:00 TUYA N][lr:0x55751] go to pre device!
[01-01 00:00:00 TUYA N][lr:0x4eb2b] reset info -> reason is 0
DPLL Unlock
[01-01 00:00:00 TUYA N][lr:0x55385] tunr on and eMode:1
[01-01 00:00:00 TUYA E][lr:0x4e8b3] upload_id [20] Dp Upload report err
[01-01 00:00:00 TUYA E][lr:0x8affb] logseq empty
[01-01 00:00:00 TUYA N][lr:0x4eb2b] reset info -> reason is 0
ota don't start or start err, can't end inform!
[01-01 00:00:00 TUYA N][lr:0xb9157] key_addr: 0x1ee000   block_sz 4096
[01-01 00:00:00 TUYA N][lr:0xb9225] get key:
0x23 0x64 0x12 0x68 0xe6 0xe6 0x2d 0x78 0x9b 0x1f 0xa0 0x80 0x85 0x1c 0xd3 0x39 
[01-01 00:00:00 TUYA N][lr:0x90f55] uni_random_init...
[01-01 00:00:00 TUYA N][lr:0x89bd9] tuya_tls_rand_init ok!
DPLL Unlock
DPLL Unlock
rate=54,channel=1,tssi_thred=0x7d,tssi_offset=0x4a4,index=64
rate=54,channel=7,tssi_thred=0x7d,tssi_offset=0x4a4,index=61
rate=54,channel=13,tssi_thred=0x7d,tssi_offset=0x4a4,index=57
rate=11,channel=1,tssi_thred=0xd0,tssi_offset=0x4a4,index=32
rate=11,channel=7,tssi_thred=0xd0,tssi_offset=0x4a4,index=30
rate=11,channel=13,tssi_thred=0xd0,tssi_offset=0x4a4,index=28
0x7
[01-01 00:00:01 TUYA N][lr:0x4c849] mf_init succ
[... 128 lines cut ...]
[01-01 00:00:07 TUYA N][lr:0x8e25d] cur bound stat:1, update bound stat:0
[01-01 00:00:07 TUYA N][lr:0xcf4df] Start Adv
[01-01 00:00:07 TUYA N][lr:0xc7dd7] __devos_init_evt success
[01-01 00:00:07 TUYA N][tuya_ble_api.c:77] dev id key: 16
6f 78 23 07 7e 3f 9f 46 4d d4 3d 57 db de 0c af 

[01-01 00:00:07 TUYA N][tuya_ble_api.c:77] adv_data 28
02 01 06 03 02 01 a2 14 16 01 a2 01 6b 65 79 6a 33 77 38 63 6d 75 74 6a 6d 77 6b 35 

[01-01 00:00:07 TUYA N][tuya_ble_api.c:77] rsp_data 30
03 09 54 59 19 ff d0 07 09 03 00 00 0c 00 c8 35 1c d4 2d 1d a4 c6 a0 2d 50 d5 de 01 2e bf 

[01-01 00:00:07 TUYA N][lr:0xcf4df] Start Adv
[01-01 00:00:07 TUYA N][lr:0x8e155] ble adv && resp changed
DPLL Unlock
cal dpll!
temp_code:45 - adc_code:309 - adc_trend:[16]:320->[17]:310
[01-01 00:00:25 TUYA N][lr:0x493db] def color:2
```

I tried to flash esphome directly via the web-serial thing, and ltchiptool gui, but I cannot do anything over UART1. Doing a CEN reset just boots the device, without any 'wait' time. The pinout also has 'P21' brought out, which doesn't seem to serve any function. Is there some trick with regards to the bootloader? Online I read everywhere 'BK7231N doesn't have a boot pin, just reset the chip'. But this hasn't been helpful in so far.

I've also tried to create the missing profile/device json's; but my gut feeling say,s, they won't be useful as we are talking 'patched SDK'.

I just want to flash 3rd party firmware, and don't mind creating the needed profiles along the way, but then how to proceed from here on?

closed 2024-12-06 by Cossid

Cossid, 2024-12-06:
Actually, the SDK line is a direct match against the other patched ones, so we know this one is also patched.  Added to the list.

oliv3r, 2024-12-06:
Thanks @Cossid and I was fearing this already :) I just hoped with the activity of cloud-cutter this was an ongoing thing (cat/mouse game). So looks like it's game over for cloudcutter for this device.

So how much timing is there to do a serial flash? less then a second it seems. Just spam reset until it catches? Oof. Is there also a DTS/DTR trick like we do on ESP's? Any pointers to what to read? The information on UART flashing seems quite light.

To your knowledge, are there BK7231N chips that have something done to them (fuses etc) that prohibits flashing/triggering flash mode?

P.S. The esphome config I had more or less figured out already

```yaml
substitutions:
  device_name: "battletron-desk-strip-3m"
  friendly_name: "Battletron Desk Strip 3 meter"
  device_description: "Battletron Desk Strip 3 meter - 3206528"

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${device_description}
  name_add_mac_suffix: true
  min_version: 2024.6.6

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    password: !secret fallback_hotspot_password

captive_portal:

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

improv_serial:

text_sensor:
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

  - platform: wifi_info
    ip_address:
      name: IP
    mac_address:
      name: MAC
    ssid:
      name: SSID
    bssid:
      name: BSSID

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

  - platform: safe_mode
    name: Safe mode
    entity_category: diagnostic

  - platform: factory_reset
    name: Factory reset
    entity_category: diagnostic

remote_receiver:
  pin:
    number: P26
    mode:
      input: true
      pullup: true
  dump: all

binary_sensor:
  - platform: gpio
    name: "Button"
    pin:
      number: P28
      inverted: true
      mode:
        input: true
        pullup: true
    on_press:
      - light.toggle: color_light

power_supply:
    id: led_power
    pin: P8

light:
  - platform: beken_spi_led_strip
    id: color_light
    name: "Color Light"
    pin: P16
    chipset: WS2812
    num_leds: 36
    rgb_order: RBG
    power_supply: led_power
    effects:
     - random:
     - pulse:
     - strobe:
     - flicker:
     - addressable_rainbow:
     - addressable_color_wipe:
     - addressable_scan:
     - addressable_twinkle:
     - addressable_random_twinkle:
     - addressable_fireworks:
     - addressable_flicker:

sensor:
  - platform: uptime
    name: "Uptime"
```

Cossid, 2024-12-06:
Technically they have an efuse, but they have never been used beyond storing an encryption key, which can already be dealt with.  We have not yet run into a device with properly exposed pins (3V3/GND/RX1/TX1) that couldn't be serial flashed.

As for the timing, LTChipTool, in my experience seems to be listening for 500ms for every 1 second frame, because it is also trying to do it's own resetting via DTS/DTR (I think), but most modules don't have the proper pins to support that, it's more of a development board thing.

Also, if you are struggling, keep your baud speed down to 115200, these devices can be a bit more tricky/touchy at higher speeds.

oliv3r, 2024-12-08:
> Technically they have an efuse, but they have never been used beyond storing an encryption key, which can already be dealt with. We have not yet run into a device with properly exposed pins (3V3/GND/RX1/TX1) that couldn't be serial flashed.

Is it written somewhere which efuses there are and what they are used for? What you speak of, is indeed where the Bootrom would expect the encryption key, or a set of fuses where the bootloader treats them as encryption keys. I read about this and that's logical. For all intents and purposes, these are 'storage' fuses if you will. There's often also configuration fuses, e.g. disable JTAG, disable UART flash or 'force secure boot' or things like that. But if this isn't really a known case, I doubt I'm suffering from it.

> 
> As for the timing, LTChipTool, in my experience seems to be listening for 500ms for every 1 second frame, because it is also trying to do it's own resetting via DTS/DTR (I think), but most modules don't have the proper pins to support that, it's more of a development board thing. So my board has brought out (as you can see) a bunch of pins, using RTS (according to the ltchiptool docs) to reset is certainly possible by tying it to CEN. This may help with the timing, but the bootloader just from start to code so fast, it's certainly less then one second. Though I would expect if I reset it manually a few times, while ltchiptool is scanning, it ought to trigger the flasher ...

> 
> Also, if you are struggling, keep your baud speed down to 115200, these devices can be a bit more tricky/touchy at higher speeds.

Thanks, I had set it to auto, but wouldn't a lower baudrate actually increase the timing? If this where a problem with reliably flashing stuff, I'd agree, but this is mostly just being able to trigger the bootloader to go into flash mode ...

Also, I'll try a different USB to UART adapter for this. While the LOG output works, who knows how finnicky things are. I have a true FTDI (without RTS) and a bus-priate, that can expose RTS on a pin. So i'll try that too.

Cossid, 2024-12-08:
> Is it written somewhere which efuses there are and what they are used for? What you speak of, is indeed where the Bootrom would expect the encryption key, or a set of fuses where the bootloader [...]

It's not well documented, but everything points to storage efuse, it's only 32 bytes, and at least half of it is an encryption key when present.  We have not witnessed a single chip so far that couldn't be booted into download mode.

> Thanks, I had set it to auto, but wouldn't a lower baudrate actually increase the timing? If this where a problem with reliably flashing stuff, I'd agree, but this is mostly just being able to [...]

The baud rate negotiation is happening much faster than any of the timing reset delays.  Lowering that should have pretty much no effect on connection time, that's milliseconds level.  It's just a matter of it happening within the reset time window, and quickly enough at the boot of the device.

kuba2k2, 2024-12-08:
It's worth mentioning that the linking is performed at 115200 baudrate, regardless of the tool setting. The chosen baudrate will only be used after linking, to speed up the actual flashing process.

oliv3r, 2024-12-08:
Using a proper FTDI cable, and resetting via CEN a few times put me into bootloader mode successfully. It took a bit of finnicking; but going well. For future archiving, I'll attach the dumped ROM here.

I tried using ltchiptool gui; but that kept segfaulting, so ended up using the cli (which is fine of course, just a steeper learning curve).

```
I: |-- Success! Chip info: BK7231N
I: Reading chip info...
I: Chip: BK7231N
I: +-----------------------+-------------------------------------+
I: | Name                  | Value                               |
I: +-----------------------+-------------------------------------+
I: | Protocol Type         | FULL                                |
I: | Chip Type             | BK7231N                             |
I: | Bootloader Type       | BK7231N 1.0.1                       |
I: | Chip ID               | 0x7231c                             |
I: | Boot Version String   | N/A                                 |
I: |                       |                                     |
I: | MAC Address           | 38:2C:E5:xx:xx:xx                   |
I: |                       |                                     |
I: | Flash ID              | EB 60 15                            |
I: | Flash Size (by ID)    | 2 MiB                               |
I: | Flash Size (detected) | 2 MiB                               |
I: |                       |                                     |
I: | Encryption Key        | 510fb093 a3cbeadc 5993a17e c7adeb03 |
I: +-----------------------+-------------------------------------+
I: |-- Finished in 2.252 s

```
[ltchiptool_bk72xx_2024-12-08_16-15-06_rom.bin.zip](https://github.com/user-attachments/files/18052779/ltchiptool_bk72xx_2024-12-08_16-15-06_rom.bin.zip)

ivohulsman, 2024-12-08:
Hey, I am currently in the process of converting the exact same product to ESPHome and happened to stumble upon this issue. I managed to get to bootloader mode, and I have sent the ESPHome Kickstart firmware, but after that the controller still behaves like it has the old firmware. Is there anything I did wrong?

oliv3r, 2024-12-08:
> Hey, I am currently in the process of converting the exact same product to ESPHome and happened to stumble upon this issue. I managed to get to bootloader mode, and I have sent the ESPHome Kickstart [...]

I didn't use the kicstarter firmware, after dumping I just flashed my esphome binary (see my yaml file above). Once I have API working; I'll send my MR to esphome-device.

Cossid, 2024-12-08:
> Hey, I am currently in the process of converting the exact same product to ESPHome and happened to stumble upon this issue. I managed to get to bootloader mode, and I have sent the ESPHome Kickstart [...]

If you have stock functionality, you did not successfully flash ESPHome-Kickstart.  Post logs.

ivohulsman, 2024-12-08:
> I didn't use the kicstarter firmware, after dumping I just flashed my esphome binary (see my yaml file above). Once I have API working; I'll send my MR to esphome-device.

Alright I'll try that, thanks! Otherwise I could wait until you have pushed a MR.

> If you have stock functionality, you did not successfully flash ESPHome-Kickstart. Post logs.

It did seem to succeed, I have attached a log.
[logfile.txt](https://github.com/user-attachments/files/18053218/logfile.txt)

Cossid, 2024-12-08:
Then define still behaving like it has the old firmware.
What logs do you get at boot on TX2?

If you can still control the device with SmartLife, then the OTA portion that was flashed somehow failed upon reboot (unsure how that could happen).  If it does not present a Kickstart AP, something didn't go correctly.

If the device simply stays lit as it previously was, that depends on the hardware configuration, as well as the default hardware configuration.

There is no way Kickstart can represent stock functionality.

Cossid, 2024-12-08:
Ah, the UG offset does appear to be incorrect, 0x132000 (T's offset) instead of 0x12A000 (N's offset).

Yeah, use the UF2 file.  Because of writing to the wrong offset, you should restore a full backup first now though.

ivohulsman, 2024-12-08:
Oops, that's an oversight on my part! Indeed, flashing the UF2 file works like a charm, thanks a lot!

oliv3r, 2024-12-08:
@ivohulsman I've updated the yaml. The number of leds was wrong, Need to set to 35 instead of 36 (what the logs provided). Probably 0 - 35 or something. With 36 leds the 'scan' would drop off the end; with 35 it bounces nicely, so I guess this must be good. Also the rgb_order was wrong; with some trial and error I have the correct one now.

This yaml file works, the API works (no idea why it didn't before).

I'll spend some time in the next few days to decode the remote control and add that too; though I debating if I'll submit that to the merge request.

ivohulsman, 2024-12-08:
> @ivohulsman I've updated the yaml. The number of leds was wrong, Need to set to 35 instead of 36 (what the logs provided). Probably 0 - 35 or something. With 36 leds the 'scan' would drop off the [...]

I did notice that the colors didn't match, but changing the number of leds fixed that. Thank you very much for all your efforts, you have made my day a lot easier!

oliv3r, 2024-12-09:
> I did notice that the colors didn't match, but changing the number of leds fixed that. Thank you very much for all your efforts, you have made my day a lot easier!

Interesting, I just looked more closely, and see that it really ought to be 36 leds. I play a bit more to get this right. Now that I can flash via esphome, It's easier to do!

ivohulsman, 2024-12-09:
> Interesting, I just looked more closely, and see that it really ought to be 36 leds. I play a bit more to get this right. Now that I can flash via esphome, It's easier to do!

You are right, somehow I did not yet notice that 1 LED is not lit up. If there is anything I can do to help please let me know.

oliv3r, 2024-12-09:
Got everything working now except https://community.home-assistant.io/t/setting-light-effect-from-a-button-binary-sensor/808061

Which should be trivial, but I have no idea yet how :)

Cossid, 2024-12-09:
You'd just call a [light.turn_on action](https://esphome.io/components/light/#light-turn-on-action) and include the `effect: <effect_name>`

oliv3r, 2024-12-10:
Perfect, how? Do you have an example? Ill try auto complete too meanwhile

Cossid, 2024-12-10:
I don't have a working example, but something along the lines of
```yaml
light:
  - platform: beken_spi_led_strip
    id: color_light
    name: "Color Light"
    pin: P16
    chipset: WS2812
    num_leds: 36
    rgb_order: GRB
    power_supply: led_power
    effects:
     - random:
         name: "random"
     - pulse:
         name: "pulse"
    # etc...
```

then in your receiver `on_press`
```yaml
on_press:
      then:
        - light.turn_on:
            id: color_light
            effect: "pulse"
```

I'm not sure if they're self-named without the explicit `name`

Cossid, 2024-12-10:
> Excellent, this was just what i was looking for and works!! And yes, you have to use the 'Default name' that is used, which annoyingly is 'fine' for things like `rainbow` and `random` because that's what the names are, however on `Color Wipe` you'd have to know this fact.

I don't think that's true, as you can have more than one effect with the same base effect type, that is why the name field exists.  You can give them any name you wish, which is why I added that in my example, I just didn't change to a unique name.

> Which works, for one second, but then HA and ESPHome have different idea's of what color to set, so I get 1 second from ESPHome which sets the color to red, and after, HA turns it to whatever color HA has self-assigned. Before it was working fine I believe (or maybe I hadn't configured HA ..) in that HA would just honor whatever ESPHome was doing.

Not sure what HA has to do with any of that.  It's possible you may have to add a delay between so they don't happen in the same loop, but I'm not certain.

oliv3r, 2024-12-11:
> I don't think that's true, as you can have more than one effect with the same base effect type, that is why the name field exists. You can give them any name you wish, which is why I added that in [...]

What I meant, is that they have (internal) default names, that don't match the id. E.g. `addressable_color_wipe` has as its name `Color Wipe`, so as a dumb user, you need to be aware that these miss-match, from a spaces and prefix point of view, because actually they are separate things. It would have been better, that the default name where something like `${id}` or something if you will.

> Not sure what HA has to do with any of that. It's possible you may have to add a delay between so they don't happen in the same loop, but I'm not certain.

Well I have:

```yaml
  - platform: remote_receiver
    name: "Color Red"
    raw:
      code: [ ... ]
    on_press:
      then:
        - light.turn_on:
            id: color_light
            effect: "none"

  - platform: remote_receiver
    name: "Color Green"
    raw:
      code: [ ... ]
    on_press:
      then:
        - light.addressable_set:
            id: color_light
            red: 0%
            green: 100%
            blue: 0%

  - platform: remote_receiver
    name: "Color Blue"
    raw:
      code: [ ... ]
    on_press:
      then:
        - light.turn_on:
            id: color_light
            effect: "none"
        - light.addressable_set:
            id: color_light
            red: 0%
            green: 0%
            blue: 100%
```

Where button 'red' works as expected and turns off the effect, green works and sets the color (affecting the effect as my per my 'issue' and blue, which would be the combination of the two doesn't. The effect gets turned off, the color is set for 1 second, and then home-assistant turns it to whatever color it thought things should be. Swapping the order obviously doesn't matter. Adding a delay might work, or should be interesting, but feels like a bug to me.

Cossid, 2024-12-11:
I still have no idea where HA is coming into play here, HA doesn't control the effect as it runs, it just starts/stops it if explicitly done, but when triggered by remote, HA shouldn't be doing anything.  I'm guessing you're just not getting the most recent state or something sent to HA.  Either way, this is far beyond the scope of CloudCutter, and any additional support should probably happen in an ESPHome issue.

oliv3r, 2024-12-11:
Absolutly! It is way way offtopic. I suppose we both just went with it :) thanks anyway, and Ill start with a bug report.

And I agree, it seems like some state isnt sent/accepted somewhere, but it must be HA that sents the color change request. Anyway, yhanks for all your help :)

oliv3r, 2024-12-13:
Just adding the Merge Request as 'final remark' here :)

https://github.com/esphome/esphome-devices/pull/947