tuya-cloudcutter/tuya-cloudcutter #745: how to get profile for Onforu 55w?

mgirgenti, 2024-12-06:
Wasn't sure where to post this. Not a bug as much as a "now what" wall to get over. I am not new to computers but relatively new to HA and ESPHome.  I do have both running fine and wanted to add smart security lighting around our property.  

I bought two of these Onforu 55w (https://www.amazon.com/dp/B0BRSQBB91) Smart LED flood lights. I turned an extra RaspPI4 into a Cloudcutter device.  It worked fine, although it took me a few runs to get the gist of it.  I did get one light cut completely.  after cutting, the flash would get stuck and not attach to cloudcutterflash.  Easy enough. I connected to A-xxx Wi-Fi and by IP, set the SSID/pass there, and poof, the flashing continued and worked perfectly.  I used ESPHome-Kickstart-v23.08.29_bk7231n and "oem_bk7231n_light_pir_ty" for ver 1.0.10.  

The light powers on and  I can see it in HA.  I tried first to just add it via ESPHome suspecting it had no config.  That was unfortunately accurate.  It adds, but has no actions, entities, etc. Wont even turn on.  

How do I use the new, unmodified device to pull the configs to so the correct YAML profile can be generated?  I don't have an android available and when I looked at Itchiptool, I keep getting a malware warning from multiple AV tools and can't run it.  I also don't want to break open the device and mess with solder points.  Needs to be OTA completely which does not seem to be an option. 

Thoughts?

Cossid, 2024-12-06:
LTChipTool + UPK2ESPHome or Kickstart's Pinscan are your only choices without doing physical inspection/tracing.  The malware warnings are false-positives, and the complete source is available to review if desired.

Warning about a blank config in ESPHome, make sure to document any secrets it may have generated if you actually deployed them, as losing them will require a serial flash.

mgirgenti, 2024-12-07:
Thanks for both your help.  I stood up a new windows VM and got the LTChipTool and UPK2ESPHome tool to run and generate a yaml file.  I did verify the pins it identified do work through the device web app. There seems to be no trigger/function or alert for movement.   Will keep playing and if I brick one, its not the end of the world.  Would love to get the actual bin file to create a proper profile, but am not set up for that yet type of connection yet.

Cossid, 2024-12-07:
You can try LightLeak to dump a bin file before flashing firmware with CloudCutter.
Also, with UPK2ESPHome, you can view the Source Data, to see if there are pins that it didn't know how to convert.  I don't think there is any conversion for PIR pins currently, but they may be present in the data.  Something like `pir_pin`

szupi-ipuzs, 2024-12-07:
In my lamp's setup, movement is triggered by P6 and the ambient light can be measured by voltage changes on P23 (adc pin).

mgirgenti, 2024-12-11:
Spent a few days and learned a bunch about ESP32, lightleak, and a borrowed old Android device.  Success.

Attached is the BIN file for the unmodified, brand new out of the box light. 

[dump_onforu_55w_Smart_Security_Light.zip](https://github.com/user-attachments/files/18088737/dump_onforu_55w_Smart_Security_Light.zip)

Cossid, 2024-12-11:
Added a profile as `Onforu BDB55WF-3 55W Motion Security Light v1.0.10`

Here is the device configuration:
```json
{
        "Jsonver": "1.0.8",
        "brightmax": 100,
        "brightmin": 5,
        "c_lv": 1,
        "c_pin": 26,
        "cagt": 20,
        "category": "0502",
        "cdsval": 2,
        "cmod": "cw",
        "colormax": 100,
        "colormaxp": 100,
        "colormin": 10,
        "colorpfun": 0,
        "crc": 41,
        "cwmaxp": 100,
        "cwtype": 0,
        "day": 157,
        "defbright": 100,
        "defcolor": "c",
        "deftemp": 0,
        "dimmod": 1,
        "dimt": 60,
        "dimval": 0,
        "dmod": 0,
        "dusk": 256,
        "evenfall": 1620,
        "evening": 2000,
        "gmkb": 60,
        "gmkg": 60,
        "gmkr": 80,
        "gmwb": 75,
        "gmwg": 70,
        "gmwr": 100,
        "keyfunc": 0,
        "lockt": 3,
        "module": "CBU",
        "night": 2180,
        "notdisturb": 0,
        "onoffmode": 0,
        "pirfreq": 20000,
        "pirhduty": 36,
        "pirin_lv": 1,
        "pirin_pin": 7,
        "pirlduty": 5,
        "pirmduty": 18,
        "pirmod": 0,
        "pirrange": 2,
        "pirsense_lv": 1,
        "pirsense_pin": 8,
        "pirwarn": 0,
        "pmemory": 1,
        "preheatt": 20,
        "prodagain": 1,
        "pwmhz": 1000,
        "remdmode": 0,
        "rgbt": 0,
        "rstbr": 50,
        "rstcor": "c",
        "rstnum": 3,
        "rsttemp": 100,
        "title20": 0,
        "trigdelay": 10,
        "trigmod": 1,
        "w_lv": 1,
        "w_pin": 24,
        "wfcfg": "spcl",
        "wfct": 3,
        "wt": 20
    }
```

And here is ESPHome yaml for the light portion:
```yaml
output:
  - platform: libretiny_pwm
    id: output_cold
    pin: P26
  - platform: libretiny_pwm
    id: output_warm
    pin: P24

light:
  - platform: cwww
    id: light_cwww
    name: Light
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    cold_white: output_cold
    warm_white: output_warm
```

PIR has pins on P7 (enable/disable sense) and P8 (binary status, requires P7 be on), you'll need to configure those to your liking with swtiches and binary sensors.

closed 2024-12-11 by Cossid

mgirgenti, 2025-04-06:
If this is the wrong place, let me know and I can move it, but several have asked how this is working. Below is the ESPHome yaml  I am using at the moment.  Not related to the yaml, but I have a 2nd light that I can't get into ESPHome yet. Stuck in the Cloud Cutter loop, but I'll figure that out another day.  Last, I am new to yaml, so this could be the worst possible way to make it work .. but it works.  Open to suggestions or optimizations. 

**Goals for the YAML:**
- Add to HA via ESPHome
- Manual on/off switch in HA
- Ability to disable/ignore motion detection
- Set the time to stay on after motion detection. Note: this is the time from the LATEST detection, not the first one.  If you have it set for 5 minutes, light comes on with first motion.  4 minutes later if motion is detected again, the 5 minutes timer is reset.
- Raise up the WiFi signal for troubleshooting methods 

**Still Missing:**
Dusk to Dawn control
- I have no idea how to access these settings/switches 
- Would be nice to control from HA is dusk-to-dawn is enabled without right separate HA automation
- Need to disable PIR during the day. At the moment, if motion is detected, it will come on even during the day 

The Yaml:

substitutions:
  device_name: pick-a-name  #change
  friendly_name: Pick a Name  #change
  comment: Onforu 55w 3-Light PIR 

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${comment}

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  password: ""

ota:
  platform: esphome
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: wifi_signal
    name: WiFi Signal
    icon: "mdi:wifi"
    update_interval: 60s
  
output:
  - platform: libretiny_pwm
    id: output_cold
    pin: P26
  - platform: libretiny_pwm
    id: output_warm
    pin: P24

light:
  - platform: cwww
    id: light_cwww
    name: none
    icon: "mdi:light-flood-down"
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    cold_white: output_cold
    warm_white: output_warm
    constant_brightness: true

switch:
  - platform: template        
    name: Enable PIR
    id: enable_PIR
    icon: "mdi:motion-sensor"
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON

number:
  - platform: template
    name: Light Off Delay (seconds)
    id: ontime_number
    icon: "mdi:timer-sync-outline"
    optimistic: true
    initial_value: 120
    min_value: 30
    max_value: 600
    step: 30
    restore_value: true

binary_sensor:
  - platform: gpio
    pin: 7
    name: PIR Sensor
    id: pir_sensor
    device_class: motion
    publish_initial_state: true
    filters:
      - delayed_off: !lambda return (id(ontime_number).state * 1000);      
    on_state:
      - if:
          condition:
           and:
            - switch.is_on: enable_PIR
            - binary_sensor.is_on: pir_sensor
          then:
            - light.turn_on: light_cwww
          else:
            - if:
                condition:
                  - switch.is_on: enable_PIR
                then:
                  - light.turn_off: light_cwww