tuya-cloudcutter/tuya-cloudcutter #746: Ledvance Tunable White B40 E14 only partially working

tommaier123, 2024-12-06:
I was able to run the script to disconnect Ledvance tunable white B40 E14 using the ledvance-tunable-white-a60-b27-bulb-v1.0.8 config (assuming the only difference is the socket).
It ran completely and generated the local keys which I was able to use to add the device to localtuya. Everything looks good and I am able to control the light.
The problem is that after restarting the light it does not work anymore.

Cossid, 2024-12-06:
Please provide complete logs of your cutting process.

tommaier123, 2024-12-07:
Here is the console output. Let me know if there is a dedicated way to generate a log, that I should be using instead.
[log.txt](https://github.com/user-attachments/files/18045898/log.txt)

Cossid, 2024-12-07:
That is what I was looking for, I just wanted to confirm it actually finished the activation process, which it appears it did.

I'm not sure there is much further advice we can give, there is no reason those keys should not continue to work.  After you reboot the device, is the device still on your local network, or has it switched back to a pairing mode?

tommaier123, 2024-12-07:
It's still on the network.
I will try using tinytuya, maybe the problem lies with localtuya.

tommaier123, 2024-12-10:
Sorry I should have been clearer.
The one that is bricked does not connect to my wifi and I can't get cloudcutter to recognize the AP (but it is there).

I would agree that my other problems are most likely due to localtuya but since tinytuya is also not working I am not so sure.

Cossid, 2025-01-09:
Closing, as CloudCutter cannot change anything here.  The issue is either your environment, or the firmware included on the device which we cannot change further.

closed 2025-01-09 by Cossid