tuya-cloudcutter/tuya-cloudcutter #750: The token cannot be intercepted

Disperion, 2024-12-11:
[+] Waiting for multicast token from app...

I can't solve the issue of getting a token

the firmware of my switch

[Tuya-Generic_W602--BLE-Light-Switch-2gang--1.3.10.bin.zip](https://github.com/user-attachments/files/18100597/Tuya-Generic_W602--BLE-Light-Switch-2gang--1.3.10.bin.zip)

You can help create a profile?

Disperion, 2024-12-11:
[image]

Cossid, 2024-12-11:
This is switch firmware version 1.3.10 which is already on the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware).  A profile cannot be added for this device and serial flashing will be your only option.

This isn't a TuyaMCU device, so the schema won't do you any good.

If you're looking for the device configuration for upk2esphome, it is

```json
{
	"bt1_lv": 0,
	"bt1_pin": 24,
	"bt2_lv": 0,
	"bt2_pin": 22,
	"bt_self_type": 1,
	"ch_cddpid1": 9,
	"ch_cddpid2": 10,
	"ch_dpid1": 1,
	"ch_dpid2": 2,
	"ch_num": 2,
	"crc": 102,
	"cyc_dpid": 43,
	"ffc_select": 0,
	"inch_dp": 44,
	"init_conf": 38,
	"jv": "110.0.0",
	"key_rst_en": 1,
	"module": "T34",
	"net_trig": 4,
	"net_type": 0,
	"netled1_lv": 0,
	"netled1_pin": 8,
	"netn_led": 1,
	"nety_led": 0,
	"on_off_cnt": 10,
	"onoff_clear_t": 10,
	"onoff_rst_m": 0,
	"onoff_rst_type": 2,
	"onoff_type": 2,
	"rand_dpid": 42,
	"remote_add_dp": 49,
	"remote_io": 6,
	"remote_list_dp": 50,
	"remote_select": 1,
	"reset_t": 5,
	"rf_width": 345,
	"rl1_lv": 1,
	"rl1_pin": 14,
	"rl2_lv": 1,
	"rl2_pin": 28,
	"series_ctrl": 0,
	"total_bt_lv": 0,
	"total_bt_pin": 10,
	"total_stat": 2,
	"zero_select": 0
}
```

closed 2024-12-11 by Cossid