tuya-cloudcutter/tuya-cloudcutter #751: Error, could not connect to SSID

akj501, 2024-12-11:
Hey,

I am running Ubuntu with an external wireless card which is capable of AP mode. When running the cloud cutter script it just seems to get stuck in a loop and then fails. The smart plug is in AP mode and I have tried 3 different WiFi adapters. Not sure what I am doing wrong? - Here is the output:

/sudo ./tuya-cloudcutter.sh -r
Building cloudcutter docker image
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  33.28MB
Step 1/10 : FROM python:3.9.18-slim-bullseye AS base
 ---> d694778260c1
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 6866e0f45502
Step 3/10 : FROM base AS python-deps
 ---> 6866e0f45502
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 47dc3d9dd75a
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> c3b41644e237
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 8bd8c2bc1715
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 60c26fe5cdf0
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 60c26fe5cdf0
Step 9/10 : COPY src /src
 ---> Using cache
 ---> e2721c53fb64
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> f29c750ed828
Successfully built f29c750ed828
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 1
Detaching requires an SSID and Password, please enter each at the following prompt
In order to provide secure logging, the values you type for your password will not show on screen
If you make a mistake, you can run the detach process again
You can also pass credentials via the -s command line option, see './tuya-cloudcutter.sh -h' for more information'
Please enter your SSID: tuya123
Please enter your Password: 
Loading options, please wait...
[?] How do you want to choose the device?: From device-profiles (i.e. custom profile) manufacturer/device name
   By manufacturer/device name
   By firmware version and name
 ► From device-profiles (i.e. custom profile)

[?] Select device profile: woox-r6080-smart-plug
   qiachip-smart-switch-v1.1.4
 ► woox-r6080-smart-plug

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 5373.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Detected app armour
This has been known to block hostapd, which is required to complete the exploit
Do you wish to stop the app armour service? [y/N] y
Unloading AppArmor profiles
AppArmour has been turned off.  You will need to manually restart it or reboot your OS for it to turn back on.
Safety checks complete.

Selected Device Slug: woox-r6080-smart-plug
Selected Profile: oem-bk7231s-rnd-switch-1.1.2-sdk-1.0.2-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "Woox-7A00", trying to connect...
Device 'wlp0s20f3' successfully activated with 'b0a72018-409a-4622-bcde-ae0fd5843634'.
Wiping NetworkManager configs
..
Found access point name: "Woox-7A00", trying to connect...
Device 'wlp0s20f3' successfully activated with 'b0a72018-409a-4622-bcde-ae0fd5843634'.
Wiping NetworkManager configs
..
Found access point name: "Woox-7A00", trying to connect...
Device 'wlp0s20f3' successfully activated with 'b0a72018-409a-4622-bcde-ae0fd5843634'.
Wiping NetworkManager configs
..
Found access point name: "Woox-7A00", trying to connect...
Device 'wlp0s20f3' successfully activated with 'b0a72018-409a-4622-bcde-ae0fd5843634'.
Wiping NetworkManager configs
..
Found access point name: "Woox-7A00", trying to connect...
Device 'wlp0s20f3' successfully activated with 'b0a72018-409a-4622-bcde-ae0fd5843634'.
Error, could not connect to SSID.
Failed to connect, please run this script again

Cossid, 2024-12-11:
Something appears to be incorrect with your NetworkManager setup.

The command being run `nmcli dev wifi connect "${AP_MATCHED_NAME}" ifname ${WIFI_ADAPTER} name "${AP_MATCHED_NAME}"` or in your case `nmcli dev wifi connect "Woox-7A00" ifname wlp0s20f3 name "Woox-7A00"` is not finishing it's connect, despite activating.

I'm guessing you need to make similar adjustments as are needed on the RPi instructions, despite not being an RPi.  https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md specifically #3, except you would use `wlp0s20f3` instead of `wlan0`

CloudCutter isn't doing anything other than making nmcli calls at this point in the script.

akj501, 2024-12-11:
To clarify here is my output for network interfaces:

/1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: wlp0s20f3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether 3c:e9:f7:f5:d3:13 brd ff:ff:ff:ff:ff:ff
3: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:45:24:7a:e8 brd ff:ff:ff:ff:ff:ff
4: wlx00c0ca58f655: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DORMANT group default qlen 1000
    link/ether 26:c0:56:a9:f8:b9 brd ff:ff:ff:ff:ff:ff permaddr 00:c0:ca:58:f6:55

wlp0s20f3 is my internal WiFi card and wlx00c0ca58f655 is my external usb WiFi card.

I have followed the instructions in the link you sent and have added the interfacesdeny for both of theses cards separately but I am getting the same result.

Cossid, 2024-12-11:
This can only be something with NetworkManager not set up correctly.
The activation message means it "connected", but the fact it starts scanning again means it never negotiated an IP.  CloudCutter will only try 5 times before giving up.

Cossid, 2024-12-11:
Additionally are you certain you have a Beken device?  Perhaps you're getting a different IP than we're expecting?

akj501, 2024-12-11:
Its a WOOX R4785 smart plug, how can I check its a Beken device?

Cossid, 2024-12-12:
There is no definitive way without opening it and checking, but you can start with https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

It looks like that device may be ESP8266 and if so, would not work with CloudCutter.  You could try Tuya-Convert, and hope it isn't also patched there.

Cossid, 2025-01-09:
Closing as stale and likely a non-compatible device.

closed 2025-01-09 by Cossid