tuya-cloudcutter/tuya-cloudcutter #753: Device dump: Zemismart Tuya WiFi Real-time Smart Energy Monitor SPM01-D2TW

ptrkp, 2024-12-17:
Device dump: Zemismart Tuya WiFi Real-time Smart Energy Monitor SPM01-D2TW firmware version 4.05
Did not work with the tuya-cloudcutter software, could not get it in slow flash mode.
Not sure if the Energy Monitor will work with this patched firmware.
[ltchiptool_bk72xx_2024-12-17_13-11-34a.zip](https://github.com/user-attachments/files/18167151/ltchiptool_bk72xx_2024-12-17_13-11-34a.zip)

Cossid, 2024-12-17:
This is actually firmware version 2.1.17, which is on the known patched list, serial flashing will remain the only option for this device.

closed 2024-12-17 by Cossid

Cossid, 2024-12-17:
4.0.5 might possibly be your TuyaMCU firmware version, as I don't think 4.0.5 exists for this firmware class, though 3.1.17 does.

Not sure what you're asking about checking values, TuyaMCU is on RX1 and TX1.

Schema id `f0bg3c` is:
```json
[{"mode":"ro","property":{"min":0,"max":999999999,"scale":2,"step":1,"type":"value"},"id":1,"type":"obj"},{"mode":"ro","id":6,"type":"raw"},{"mode":"ro","property":{"type":"bitmap","maxlen":17},"id":9,"type":"obj"},{"mode":"rw","property":{"type":"bool"},"id":12,"type":"obj"},{"mode":"rw","id":17,"type":"raw"},{"mode":"rw","id":18,"type":"raw"},{"mode":"ro","property":{"type":"string","maxlen":255},"id":19,"type":"obj"},{"mode":"ro","property":{"min":0,"max":999999999,"scale":2,"step":1,"type":"value"},"id":23,"type":"obj"},{"mode":"ro","property":{"min":0,"max":9999,"scale":2,"step":1,"type":"value"},"id":32,"type":"obj"},{"mode":"rw","property":{"range":["online","offline"],"type":"enum"},"id":35,"type":"obj"},{"mode":"rw","property":{"min":-999,"max":999,"scale":2,"step":1,"type":"value"},"id":50,"type":"obj"},{"mode":"ro","property":{"min":0,"max":5000,"scale":1,"step":1,"type":"value"},"id":102,"type":"obj"},{"mode":"ro","property":{"min":-100000,"max":100000,"scale":3,"step":1,"type":"value"},"id":103,"type":"obj"},{"mode":"ro","property":{"min":-20000,"max":20000,"scale":3,"step":1,"type":"value"},"id":104,"type":"obj"}]
```

ptrkp, 2024-12-17:
That is what I saw in in the Tuya App.
The values of energy, current, voltage, for now no values are showing up in esphome