tuya-cloudcutter/tuya-cloudcutter #755: issue trying to flash tuya version 1.3.20

yonubear, 2024-12-18:
im trying to flash a bulb that has v1.3.20 but I am lost I am told it is the same chip as a nous gu10  but all I get no matter what I choose is on the second Ap scan that it resulted in no exploid so before I start trying to reflash via soldering I thought I would ask here

Cossid, 2024-12-18:
When an exact profile match doesn't exist, you should use by firmware version / name option, but there was not a 1.3.20 option.  1.3.19 and 1.3.21 were common for lights.

I was able to locate a patch and created a profile for 1.3.20, so you should now be able to select that via by firmware version / name.

closed 2024-12-18 by Cossid

yonubear, 2024-12-18:
Thank you i will try that and pist back if it works.