tuya-cloudcutter/tuya-cloudcutter #756: Device does not want to connect to the cloudcutterflash AP

sanyadez, 2024-12-18:
Connection does not occur for more than 10 minutes. The flashing process got stuck on this the first time and any other attempt gives the same thing. Btw, first bulb has been flashed w/o any problems, another 2 have this issue. Logs are added. 

Hardware:
 - Raspberry pi zero
 - Unbranded bulb with the fw 2.9.6

Selected Device Slug: woox-r5111-ceiling-light-v2.9.6
Selected Profile: oem-bk7231s-light-ty-2.9.6-sdk-1.0.2-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

Any another ways to flash the bulbs through the air?

`================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP

Found access point name: "A-F3CB", trying to connect...
Device 'wlan0' successfully activated with '2c7bacc8-edbe-44d3-98bb-16fe0b78d73e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/YuEYYfW3Rp6b.deviceconfig
Saved device config in /work/configured-devices/YuEYYfW3Rp6b.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.......
Found access point name: "A-F3CB", trying to connect...
Device 'wlan0' successfully activated with '6b07308a-03c9-4ab7-ab9e-facbe3641bee'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr b8:27:eb:d0:f7:af and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 

`

Cossid, 2024-12-18:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

sanyadez, 2024-12-18:
Yep, thanks. That works. Awesome and quick answer

closed 2024-12-18 by sanyadez