tuya-cloudcutter/tuya-cloudcutter #757: tuya fan firmware 1.2.2

yonubear, 2024-12-19:
i fugured i would try my luck again I am trying to flash a tuya based ceiling fan that has firmware 1.2.2 and mcu 1.2.2 it tells me when I try that it doesn't result in a success and exits I am using the option that says version 1.2.2 light is there a different  option I should try if need be I can attempt to do a dump of the existing flash.

Here are photos of the board and model info if it helps

[image]
[image]

Cossid, 2024-12-19:
Fan and light are different firmwares (even if the fan has a light).  You'll probably need to dump this one.

yonubear, 2024-12-19:
thank you i kind of figured it would be different but was unsure how to see if I can figure out the dumping of the firmware as this will be my first time having to lol

Cossid, 2024-12-19:
I should also add, that is an unlabeled module, so it is no guarantee it is Beken.  It could also be Realtek or LightningSemi.
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out is the only guidance we have, but it is far from perfect.

yonubear, 2024-12-19:
I will give that a rea and the second antenna is for the RF remote I would assume and the date on the box is 2022 so I would love love to find out it has a 8266 just because it might be simpler lol

yonubear, 2024-12-19:
based on mac address is is either a bekten or a realtek

yonubear, 2024-12-21:
well I tried lightleak to dump the chip and all got was that the device failed to stop responding so I guess I will now pull the chip and see what happens

Cossid, 2025-01-05:
Closing until the chip can be positively identified and a dump provided.  If you are able to provide a dump, please attach it and re-open this issue.

closed 2025-01-05 by Cossid