tuya-cloudcutter/tuya-cloudcutter #759: How to flash 55W CCT APFL

Hoerli1337, 2024-12-21:
Hey!
I have a Briloner lamp which has a “Meko LED Driver” with the model designation MKP060C1450BLTG-1 installed as the LED driver.
(This one: https://www.amazon.de/BRILONER-Fernbedienung-Wohnzimmerlampe-Schlafzimmerlampe-Deckenbeleuchtung/dp/B08VRXVT37)
In the Tuya app, the lamp is initially called “55W CCT APFL”.
The following firmware is installed:
Main module: V3.1.4
MCU module: V1.0.0

Which profile can I use for flashing?
Unfortunately I can't find anything with the name of the lamp.

If it helps, I also have a “Virtual ID”.

Cossid, 2024-12-21:
3.1.4 is not a currently supported firmware, so would require a dump.

Have to opened the driver box yet to verify which wifi module is inside?  Based on that unique firmware version, there is a decent chance this is an older ESP or Realtek module, and wouldn't be supported by CloudCutter.

Hoerli1337, 2024-12-21:
I have not yet opened the box.
When I search for the Wifi MAC on the net, this is what comes up:

**The OUI 84:CC:A8 was given by the IEEE to**
Espressif Inc.
Room 204, Building 2, 690 Bibo Rd, Pudong New Area
Shanghai  Shanghai  201203
CN

Cossid, 2024-12-21:
That means the module is made by Espressif, which means it is most likely ESP8266, and would never work with CloudCutter.

If Tuya-Convert doesn't work, you'll probably need to serial flash.  Closing as it is outside the realm of CloudCutter.

closed 2024-12-21 by Cossid

Hoerli1337, 2024-12-21:
Okay thanks for the info
:(