tuya-cloudcutter/tuya-cloudcutter #760: Unsupported new Device - is it flashable anymore (Firmware 1

Aaroneisele55, 2024-12-22:


Cossid, 2024-12-22:
According to https://templates.blakadder.com/teckin_SP22.html that device was only available with either ESP or Realtek modules, neither of which work with CloudCutter.

closed 2024-12-22 by Cossid

Aaroneisele55, 2024-12-22:
> According to https://templates.blakadder.com/teckin_SP22.html that device was only available with either ESP or Realtek modules, neither of which work with CloudCutter.

I already tried to tuya-convert it, but it has that damn new Realtek chip, not an ESP....
So it is completely impossible to flash it without opening it?

Cossid, 2024-12-22:
More than likely, yes.  There are known vulnerabilities for Realtek, but no working exploits.

Cossid, 2025-12-09:
RTL8710BN is now supported for devices that are vulnerable, however, we don't know about firmware 1.5.0 for this device, so a dump would be needed to verify/support.  We have only seen firmware 1.0.0 for this device, which is not currently vulnerable.