tuya-cloudcutter/tuya-cloudcutter #761: LSC GU10 RGBCT bulb v1.3.31

Romain-GUILLEMOT, 2024-12-30:
Hello,

I have an LSC GU10 bulb running firmware version 1.3.21. Unfortunately, this bulb doesn’t seem to be included in the pre-made configurations for LSC devices.

Could you please guide me on the steps I need to take with my bulb to create a custom configuration? I have a UART adapter, but I’m not sure exactly what information I need to retrieve to create the config.

Thank you!

Cossid, 2024-12-30:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

& 

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#i-have-flashed-a-3rd-party-firmware-how-can-i-configure-my-device

closed 2024-12-30 by Cossid

Romain-GUILLEMOT, 2024-12-30:
I spent several hours retrieving the device configuration via UART, and after creating the default profile, I realized that one of the three bulbs had a different firmware version. Of course, by sheer luck, I worked on the only bulb with a different version—also the only one that already had a profile on Cloudcutter.

As for the other bulbs running firmware version 1.3.21, the CMPAR38-120W-AL-9ACK CB8P RGBCT Bulb 1.3.21 profile seems to work perfectly with these LSC bulbs.