tuya-cloudcutter/tuya-cloudcutter #762: Device stalled in "A-####" AP mode

VACIndustries, 2025-01-01:
Was trying to get a KMC Power Strip cut/flashed for ESPHome...and it seemed like the flashing went fine, but when waiting for the AP to come up, it never did.  Tried "resetting" the device like you do when cutting (hold button for 6 seconds, then again); and it brought up the AP but in "A-####" mode instead of the Kickstart (ESPHome) like I was expected to find; here is the dump of that portion of the run:
```

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr e4:5f:01:c1:XX:XX and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
If your device gets stuck here with no progress after several (at least two) minutes, see https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do for additional steps
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.property.save
Processing endpoint tuya.device.uuid.pskkey.get
Processing endpoint atop.online.debug.log
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dynamic.config.ack
Processing endpoint tuya.device.timer.count
Processing endpoint tuya.device.upgrade.silent.get
Processing endpoint tuya.device.upgrade.status.update
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 1%
Firmware update progress: 1%
Firmware update progress: 1%
Firmware update progress: 1%
Firmware update progress: 1%
Firmware update progress: 1%
Firmware update progress: 2%
Firmware update progress: 2%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=0-561887/561888
Firmware update progress: 2%
Firmware update progress: 2%
Firmware update progress: 2%
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 2%
Firmware update progress: 5%
Firmware update progress: 5%
Firmware update progress: 5%
Firmware update progress: 5%
Firmware update progress: 5%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=14247-561887/561888
Firmware update progress: 5%
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 5%
Firmware update progress: 11%
Firmware update progress: 15%
Firmware update progress: 15%
Firmware update progress: 15%
Firmware update progress: 15%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=32819-561887/561888
Firmware update progress: 15%
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 15%
Firmware update progress: 19%
Firmware update progress: 20%
Firmware update progress: 20%
Firmware update progress: 22%
Firmware update progress: 25%
Firmware update progress: 25%
Firmware update progress: 26%
Firmware update progress: 26%
Firmware update progress: 26%
Firmware update progress: 26%
Firmware update progress: 30%
Firmware update progress: 32%
Firmware update progress: 32%
Firmware update progress: 32%
Firmware update progress: 37%
Firmware update progress: 37%
Firmware update progress: 37%
Firmware update progress: 37%
Firmware update progress: 37%
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 37%
Firmware update progress: 44%
Firmware update progress: 49%
Firmware update progress: 59%
Firmware update progress: 63%
Firmware update progress: 66%
Firmware update progress: 67%
Processing endpoint tuya.device.timer.astronomical.list
Firmware update progress: 68%
Firmware update progress: 69%
Firmware update progress: 70%
Firmware update progress: 71%
Firmware update progress: 74%
Firmware update progress: 77%
Firmware update progress: 77%
Firmware update progress: 78%
Firmware update progress: 82%
Firmware update progress: 83%
Firmware update progress: 84%
Firmware update progress: 84%
Firmware update progress: 85%
Firmware update progress: 85%
Firmware update progress: 85%
Firmware update progress: 85%
Firmware update progress: 85%
Firmware update progress: 85%
Firmware update progress: 85%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=210123-561887/561888
Firmware update progress: 85%
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Firmware update progress: 85%
Firmware update progress: 87%
Firmware update progress: 90%
Firmware update progress: 90%
Firmware update progress: 90%
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=481274-561887/561888
Firmware update progress: 94%
Firmware update progress: 95%
Firmware file has been sent and MQTT reported a progress of nearly complete.  Waiting 15 seconds to ensure flashing completes.
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=87891-561887/561888
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=543646-561887/561888
Flashing should be complete.  It takes about 15 seconds for the device to reboot and verify the flash was valid.
Please wait about 30 seconds then look for signs of activity from the firmware you supplied (either watch for AP mode or check if it joined your network).
Device MAC address: a8:80:55:07:XX:XX
```

Cossid, 2025-01-01:
Looks like you have a poor connection, and it tried to resume multiple times as a result.  Just move the device closer and do the same process again.

There isn't a "complete" status sent, so CloudCutter assumes anything > 90% completed, based on the messages that typically do send.  Alternatively, you can increase the timeout towards the end with the `-t` parameter, default is 15 seconds, if your connection remains slow like that, it may need more time.

closed 2025-01-01 by Cossid

VACIndustries, 2025-01-01:
Thanks @Cossid for the really quick response.  You're most definitely right about the distance to the device needing to be closer...  though when I rerun through the process again, it's looking for the Tuya AP (`Scanning for open Tuya SmartLife AP`).   Is there a way to ensure it's looking for the new access point of `A-####` now?

Cossid, 2025-01-01:
It will pick up the `A-` prefix, it is actually just scanning for anything open ending with `-XXXX`

VACIndustries, 2025-01-02:
Thanks again for your assistance @Cossid -- got it functioning