tuya-cloudcutter/tuya-cloudcutter #763: Device Dump - MOES WS-US-RF

Gadorach, 2025-01-02:
[ltchiptool_bk72xx_3-gang_WS-US-RF.zip](https://github.com/user-attachments/files/18287467/ltchiptool_bk72xx_3-gang_WS-US-RF.zip)

MOES Home 3-Gang WiFi Smart Switch. Contains a BK7231T (WB2S module, reports as BK7231S). Cannot download device schema from Tuya as it likely has a permissions error. Extracted schema from ltchiptool is as follows:
```
{
    "gw_bi": {
        "uuid": "ec7a9074e6e05929",
        "psk_key": "BFF9EH8AnmeQnB1gQXU1blD2aE1wK7u3rFD0z",
        "auth_key": "3jS3z32aAC6MkreJTIGTJiCWGLv43cMe",
        "ap_ssid": "SmartLife",
        "ap_passwd": null,
        "country_code": "CN",
        "prod_test": false
    },
    "user_param_key": {
        "crc": 11,
        "wifi_bdrate": 9600,
        "wifi_cofig_ver": "1.0.0"
    },
    "gw_wsm": {
        "nc_tp": 3,
        "ssid": "VmVydGV4SG90c3BvdA==",
        "passwd": "bG1hb251Yml3aW4=",
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 2,
        "token": "OanZckiZ",
        "region": "AZ",
        "reg_key": "Q1oQ"
    },
    "tls_ca_cnt": 0,
    "gw_di": {
        "abi": 0,
        "id": "eb6c30708cfc453ce70rea",
        "swv": "1.0.9",
        "bv": "30.06",
        "pv": "2.2",
        "lpv": "3.3",
        "pk": "m4viuhaiez8qjswh",
        "firmk": "keyvgeknmyujaf4m",
        "cadv": "1.0.2",
        "cdv": "1.0.0",
        "dev_swv": "3.1.4",
        "s_id": "0000012b1i",
        "dtp": 1,
        "sync": 0,
        "attr_num": 0,
        "mst_tp_0": 0,
        "mst_ver_0": null,
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_tp_3": 0,
        "mst_ver_3": null,
        "dminfo_name": null,
        "dminfo_code": null
    },
    "gw_ai": {
        "key": "c45113c2751f4345",
        "lckey": "8e376c4ce238f332",
        "h_url": "http://a.tuyaus.com/d.json",
        "hs_url": null,
        "hs_psk": "https://a3.tuyaus.com/d.json",
        "mqs_url": null,
        "mq_url": "m2.tuyaus.com:1883",
        "ai_sp": null,
        "mq_psk": "m2.tuyaus.com:8886",
        "time_z": "-04:00",
        "s_time_z": "[[1741500000,1762059600],[1772949600,1793509200]]",
        "wx_app_id": null,
        "wx_uuid": null,
        "dy_tls_m": 0
    },
    "timer_arr": {
        "lastFetchTime": 0,
        "cnt": 0
    },
    "0000012b1i": [
        {
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 1,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 2,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 3,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 7,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 8,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 9,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "range": [
                    "0",
                    "1",
                    "2"
                ],
                "type": "enum"
            },
            "id": 14,
            "type": "obj"
        },
        {
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 16,
            "type": "obj"
        }
    ],
    "mf_test_close": true,
    "em_sys_env": "BK7231S_2M"
}
```

Expected Layout for Tuya MCU (Holtek BS86D12C in this case for touch controller):

-------------------------------
Multi-gang config. Sold in multiple configurations. Switches extrapolated from 3-gang variant.
-------------------------------
```
id: 1 //Switch ON/OFF 1
id: 2 //Switch ON/OFF 2
id: 3 //Switch ON/OFF 3
id: 4 //Switch ON/OFF 4
id: 5 //Switch ON/OFF 5
id: 6 //Switch ON/OFF 6

id: 7  //Auto-Off Timer (Switch 1)
 - Min 0s, Max 86400s (24h)
id: 8  //Auto-Off Timer (Switch 2)
 - Min 0s, Max 86400s (24h)
id: 9  //Auto-Off Timer (Switch 3)
 - Min 0s, Max 86400s (24h)
id: 10 //Auto-Off Timer (Switch 4)
 - Min 0s, Max 86400s (24h)
id: 11 //Auto-Off Timer (Switch 5)
 - Min 0s, Max 86400s (24h)
id: 12 //Auto-Off Timer (Switch 6)
 - Min 0s, Max 86400s (24h)

id: 14 //Indicator Light Status
 - 0 = OFF (Button Light always OFF)
 - 1 = ON/OFF Status (Button Light ON when Switch ON)
 - 2 = Switch Position (Button Light ON when Switch OFF)

id: 16 //Restart Status? Not sure what this does. Noted as "Restart Memory". Likely clears the Tuya device memory to be set up again.
```
-------------------------------
Possible valid ESPHome Config:
-------------------------------

```
esphome:
  name: "moes-smart-switch"
  name_add_mac_suffix: true
  friendly_name: "MOES WS-US-RF Smart Switch"

bk72xx:
  board: wb2s

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_enc_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret ap_ssid
    password: !secret ap_password

captive_portal:

web_server:
  
logger:

mdns:
  
button:
  - platform: restart
    name: Restart

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: uptime
    name: Uptime

uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

tuya:
  # DPIDs guessed based on device info from 0000012b1i config

light:
  - platform: tuya
    name: "Switch 1"
    switch_datapoint: 1       # Controls power on/off
  - platform: tuya
    name: "Switch 2"
    switch_datapoint: 2       # Controls power on/off
  - platform: tuya
    name: "Switch 3"
    switch_datapoint: 3       # Controls power on/off
#  - platform: tuya
#    name: "Switch 4"
#    switch_datapoint: 4       # Controls power on/off
#  - platform: tuya
#    name: "Switch 5"
#    switch_datapoint: 5       # Controls power on/off
#  - platform: tuya
#    name: "Switch 6"
#    switch_datapoint: 6       # Controls power on/off

number:
  - platform: tuya
    number_datapoint: 7
    name: "Switch 1 Timer (min)"
    unit_of_measurement: min
    mode: box
    min_value: 0
    max_value: 1440
    multiply: 60
    step: 1
  - platform: tuya
    number_datapoint: 8
    name: "Switch 2 Timer (min)"
    unit_of_measurement: min
    mode: box
    min_value: 0
    max_value: 1440
    multiply: 60
    step: 1
  - platform: tuya
    number_datapoint: 9
    name: "Switch 3 Timer (min)"
    unit_of_measurement: min
    mode: box
    min_value: 0
    max_value: 1440
    multiply: 60
    step: 1
#  - platform: tuya
#    number_datapoint: 10
#    name: "Switch 4 Timer (min)"
#    unit_of_measurement: min
#	   mode: box
#    min_value: 0
#    max_value: 1440
#	   multiply: 60
#	   step: 1
#  - platform: tuya
#    number_datapoint: 11
#    name: "Switch 5 Timer (min)"
#	   unit_of_measurement: min
#	   mode: box
#    min_value: 0
#    max_value: 1440
#	   multiply: 60
#	   step: 1
#  - platform: tuya
#    number_datapoint: 12
#    name: "Switch 6 Timer (min)"
#	   unit_of_measurement: min
#	   mode: box
#    min_value: 0
#    max_value: 1440
#	   multiply: 60
#	   step: 1

select:
  - platform: tuya
    enum_datapoint: 14
    name: "Indicator Light Mode"
    optimistic: true
    options:
      0: Always Off
      1: On when Switch On
      2: On when Switch Off

switch:
  - platform: tuya
    switch_datapoint: 16
    name: Factory Reset
    internal: true
    on_turn_on:
      then:
        - switch.turn_on: reset
        
  - platform: factory_reset
    id: reset
    internal: true
```

Gadorach, 2025-01-02:
I believe it to relate to this model.

https://moeshouse.com/products/us-rf433-wifi-wall-touch-switch-no-neutral?pr_prod_strat=e5_desc&pr_rec_id=eb16a3852&pr_rec_pid=4672084541521&pr_ref_pid=6819801399377&pr_seq=uniform

Note that I bought these nearly 8 years ago, so I can't be sure that they are still made with the same parts. I doubt anything changed though.

Gadorach, 2025-01-02:
If anyone is trying to rip these, note that you will need to lift pin 21 (VDD) on the Holtek BS86D12C to disable it first, or it will overtake the serial communications. Unfortunately it does not have a reset pin, so you can't hold it down to reset, and the traces go directly to the WB2S module. There is no ability to just remove a resistor to disable it, or pull a reset line to ground, for example.

Cossid, 2025-01-02:
Added as `MOES WS-US-RF 3-Gang Switch v1.0.9`

Here is the data model from Tuya with the codes, names need to be run through a translator, but are usually similar to the codes.

```json
[{
        "abilityId": 1,
        "accessMode": "rw",
        "code": "switch_1",
        "description": "",
        "extensions": {
            "iconName": "icon-power",
            "attribute": "21"
        },
        "name": "开关",
        "typeSpec": {
            "type": "bool"
        }
    }, {
        "abilityId": 2,
        "accessMode": "rw",
        "code": "switch_2",
        "description": "",
        "extensions": {
            "iconName": "icon-power",
            "attribute": "21"
        },
        "name": "开关 2",
        "typeSpec": {
            "type": "bool"
        }
    }, {
        "abilityId": 3,
        "accessMode": "rw",
        "code": "switch_3",
        "description": "",
        "extensions": {
            "iconName": "icon-power",
            "attribute": "21"
        },
        "name": "开关 3",
        "typeSpec": {
            "type": "bool"
        }
    }, {
        "abilityId": 7,
        "accessMode": "rw",
        "code": "countdown_1",
        "description": "",
        "extensions": {
            "iconName": "icon-dp_time2"
        },
        "name": "开关1倒计时",
        "typeSpec": {
            "type": "value",
            "max": 86400,
            "min": 0,
            "scale": 0,
            "step": 1,
            "unit": "s"
        }
    }, {
        "abilityId": 8,
        "accessMode": "rw",
        "code": "countdown_2",
        "description": "",
        "extensions": {
            "iconName": "icon-dp_time2"
        },
        "name": "开关2倒计时",
        "typeSpec": {
            "type": "value",
            "max": 86400,
            "min": 0,
            "scale": 0,
            "step": 1,
            "unit": "s"
        }
    }, {
        "abilityId": 9,
        "accessMode": "rw",
        "code": "countdown_3",
        "description": "",
        "extensions": {
            "iconName": "icon-dp_time2"
        },
        "name": "开关3倒计时",
        "typeSpec": {
            "type": "value",
            "max": 86400,
            "min": 0,
            "scale": 0,
            "step": 1,
            "unit": "s"
        }
    }, {
        "abilityId": 14,
        "accessMode": "rw",
        "code": "relay_status",
        "description": "",
        "extensions": {
            "iconName": "icon-dp_circle"
        },
        "name": "上电状态设置",
        "typeSpec": {
            "type": "enum",
            "range": ["0", "1", "2"]
        }
    }, {
        "abilityId": 16,
        "accessMode": "rw",
        "code": "switch_backlight",
        "description": "",
        "extensions": {
            "iconName": "icon-dp_light",
            "attribute": "16"
        },
        "name": "背光开关",
        "typeSpec": {
            "type": "bool"
        }
    }
]

```

closed 2025-01-02 by Cossid